carla-simulator/carla #8742: [Possible Bug] Significant Deviation in Vehicle Behavior When Replaying Control Data from Traffic Manager

AOOOOOA, 2025-03-11:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: Carla 0.9.14 
  * Platform: Ubuntu 18 
  * Python version: Python 3.7 
  * GPU: Nvidia 3080
  * GPU Drivers: 525.105.17 

**Describe the bug**

 I conducted an experiment to compare vehicle behavior between two runs:
**1st Run**: Vehicles were controlled using the Traffic Manager, and I recorded the control data (e.g., throttle, brake, and steering commands) for each vehicle.
**2nd Run**: I disabled the Traffic Manager, spawned vehicles of the same type at the same locations, and replayed the recorded control commands from the 1st run.

**Expectation**: Since the same control commands were applied in an identical environment, the vehicle states (position, velocity, acceleration) in the 2nd run should match those in the 1st run.

**Observation**: There are noticeable discrepancies between the two runs. Vehicle behavior deviates to varying degrees, and in some cases, the deviation is extreme. Below is a video showing a significant trajectory error during the 2nd run, where a vehicle deviates so much that it crashes.

The Carla simulation is set to synchronous mode with a fixed time step of 0.05, and the physics substep is configured to 10 with an interval of 0.05.

https://github.com/user-attachments/assets/570d56d0-701a-439a-9d26-9aa0980afcce

 The figure is the vehicle's error in the video between two runs. 
[image]

**Steps to reproduce**

To reproduce the issue, you can follow these steps to record control data using the Traffic Manager (TM) and replay it with the recorded commands. I have attached the relevant code in a zip file for your convenience.

**Steps to Reproduce:**
**1st Run:** Use the Traffic Manager to control vehicles and record their control data.

python TM_1st.py --town Town02 -n 5 --filterv vehicle.toyota.prius --record-path TM_control_data.json
--town: Specifies the map (e.g., Town02).
-n: Number of vehicles to spawn (e.g., 5).
--filterv: Filters the vehicle type (e.g., vehicle.toyota.prius).
--record-path: Path to save the recorded control data.

**2nd Run:** Replay the recorded control commands without using the Traffic Manager.

python Replay_recorded_command.py --town Town02 --replay-path TM_control_data.json --output-path replay_result.json
--replay-path: Path to the recorded control data from the 1st run.
--output-path: Path to save the replayed results.

**Analyze the Results:** Optionally, you can use the provided error analyzer to compare the high-load (recorded) and low-load (replay) results.

python error_analyzer.py --high-load TM_control_data.json --low-load replay_result.json --output output_folder/
--high-load: Path to the recorded control data.
--low-load: Path to the replayed results.
--output: Folder to save the analysis results.

**Additional Notes:**
**Reproducibility:** The extreme trajectory deviation is not guaranteed to occur in every single run. To increase the chances of observing the issue, you can use the provided script for repeated tests:

python multiple_run_test.py --config config/config.json
Configure the Carla Path: Before running, modify the CARLA_PATH variable in multiple_run_test.py to match your local Carla installation path.

**Default Behavior:** By default, this script runs the test 10 times (you can adjust the number of iterations as needed). It outputs the results of each run, allowing you to identify cases with extreme errors for further analysis.

**Expected behavior**
Using the same control commands, the same vehicle type, and the same running environment, the vehicle's performance is expected to be identical. I tested this by running the same recorded control commands twice, and as expected, the vehicle's performance was deterministic and consistent across both runs.

However, when using the Traffic Manager (TM) during the first run and then replaying the recorded commands in the second run, deviations consistently occur.

**Logs**
Paste the server and client logs. Server ones can be found at:
  * Unreal Engine 4: _Unreal/CarlaUE4/Saved/Logs_
  * Unreal Engine 5: _Unreal/CarlaUnreal/Saved/Logs_

**Scripts**

[TM_replay.zip](https://github.com/user-attachments/files/19186785/TM_replay.zip)

AOOOOOA, 2025-03-13:
@Blyron  Hi Blyron, could you please help with this? Thanks!

Blyron, 2025-03-13:
We are trying to do our best. We will get here when we can

AOOOOOA, 2025-03-14:
> We are trying to do our best. We will get here when we can

Hi Blyron, thank you for your efforts! However, the issue with the Traffic Manager may point to a critical accuracy concern in Carla. Addressing this promptly would be essential to ensure the reliability and credibility of the framework. Could you kindly prioritize this and provide an update?

Blyron, 2025-03-14:
You could get premium support to look fast to this issue. Meanwhile, we are aware of this. Thanks for your explanation of the issue.

AOOOOOA, 2025-03-14:
> You could get premium support to look fast to this issue. Meanwhile, we are aware of this. Thanks for your explanation of the issue.

 
No, actually what I mean is that resolving this issue will enhance the reliability and credibility of Carla itself. I also noticed that some users have submitted similar issues in the past. 
[https://github.com/carla-simulator/carla/issues/6477](url) 
[https://github.com/carla-simulator/carla/issues/6478](url)

That said, I have also submitted a request to Carla's premium support. I’d greatly appreciate your time and effort in looking into this problem.

Thank you!

AOOOOOA, 2025-03-19:
@Blyron  Hi Blyron, do you have time for this?

AOOOOOA, 2025-04-23:
Hi @Blyron , just following up on this issue. Do you have any updates or an estimated timeline for resolution? Let me know if there’s anything I can assist with. Thanks!

Blyron, 2025-04-23:
No, we do not have estimates

Blyron, 2025-04-23:
I aware of it. 

Carla has private contracts to sustain the project which has more priority and other tasks that have more priority than solving this issue. We are aware of this issue. We are solving the issues as much as we can. Some issues are easy to solve, others like this, are not. 

On the other hand, you could try to sign a contract with us so we can fix this issue with maximum priority.

If you want to, we could have business call to agree a contract to fix this issue.

AOOOOOA, 2025-04-24:
We’d be happy to contribute a PR to help resolve the issue — but first, we’d appreciate it if you could confirm whether this behavior is expected by design, or if it’s considered a bug or unintended. This clarification will help us decide whether to move forward with a fix.

Regarding the contract proposal, thank you for the offer. We're independent developers and not currently considering any business agreements. 

Thanks again for your continuous work on Carla. Looking forward to your clarification.

AOOOOOA, 2025-05-16:
Hi Blyron, @Blyron

I have some new findings regarding the root cause of the problem.

I inserted logging statements into the underlying C++ code and recompiled it to run again. I noticed that the gear values within the control command appear inconsistent in both the Client side and the Server side.

During two runs (the first using the Traffic Manager and the second applying the recorded control command from the first run), the gear value obtained from the Client-side Get_Control function does not match the value from the Server-side Apply Control function. I confirmed that I recorded the gear value using the Python API in the first run and re-applied it to the vehicle in the second run. This suggests that the gear value for the vehicle cannot be correctly set or retrieved within CARLA.

[image]

[image]

Additionally, I compared the gear values from both the client and server sides during a single run, and I have included the comparison figures below.

[image]

Could you please follow up on this issue? Thanks!