carla-simulator/carla #8743: Instance Segmentation Camera still sees environment objects after disabling them

KJ-Chang, 2025-03-12:
I tried to disable environment objects (e.g., fences) using `enable_environment_objects(False)`, and it worked for the RGB camera. However, the **Instance Segmentation Camera** still detects and displays them.

```python
object_labels = [carla.CityObjectLabel.Car, 
                            carla.CityObjectLabel.Truck, 
                            carla.CityObjectLabel.Bus, 
                            carla.CityObjectLabel.Pedestrians,
                            carla.CityObjectLabel.Bicycle,
                            carla.CityObjectLabel.Rider,
                            carla.CityObjectLabel.Train,
                            carla.CityObjectLabel.Motorcycle,
                            carla.CityObjectLabel.Fences # 圍欄
                            ]
            obj_ids = []
            for label in object_labels:
                for obj in sim_world.get_environment_objects(label):
                    obj_ids.append(obj.id)
            # 事先移除地圖原先上的物件
            sim_world.enable_environment_objects(obj_ids, False)
```

[image]

KJ-Chang, 2025-03-12:
It's normal at Town10HD.

before
[image]
after
[image]

Blyron, 2025-03-12:
Please use the provided issue templates.

closed 2025-03-12 by Blyron