carla-simulator/carla #8745: Semantic segmentation camera cannot recognize user-defined landscape

yuxinhi, 2025-03-12:
* CARLA version: 0.9.14
  * Platform: ubuntu 20.04
  * GPU: NVIDIA GeForce RTX 4080

When using the source-compiled version of Carla, it was discovered that the semantic camera fails to recognize the Landscape created using the terrain editing tool. The Landscape appears as black in the semantic camera, whereas the default maps in Carla recognize the Landscape with the "Other" semantic label.

I created a custom Landscape on Carla's built-in BaseMap, but the semantic camera still fails to recognize it. Below are the recognition results. What modifications should I make to ensure that the user-defined Landscape is recognized by the semantic camera with a custom tag or the "Other" tag?

**Screenshots**
[image]

Blyron, 2025-03-12:
Semantic segmentation are recognised in latest ue4-dev branch, sorry for inconveniences

closed 2025-03-12 by Blyron

yuxinhi, 2025-03-14:
After using the command `git clone -b ue4-dev https://github.com/carla-simulator/carla` to clone the carla ue4-dev repository on Ubuntu 20.04, I encountered the following error when compiling the ue4-dev environment with the make PythonAPI command. 

[image]

I tried the method from #6974, replacing and modifying the original FileTransfer.h, DVSCamera.h, and Util/BuildTools/Setup.sh files, but the issue still persists. I know that the ue4-dev is a dev branch based on Carla 0.9.15. Is this version supported for building from source on Ubuntu 20.04？How can I achieve it or do I need to upgrade my system to Ubuntu 22.04? 

Thanks you for your reply.