carla-simulator/carla #8747: 0.10.0 make launch error on ubuntu 22.04 due to Zen Storage and DerivedDataCache

CStubenvoll, 2025-03-12:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10.0
  * Platform: ubuntu 22.04

**Describe the bug**
I followed the Extended Build Instructions for the ue5 version of Carla for ubuntu 22.04, since I had some issues when running the setup script. The python API built sucessfully. 
Next step is to launch the editor with the following command: `cmake --build Build/Release/ --target launch` (the documentation wrongly says "Linux-Release" here but it is "Release" btw). But the launch always fails with errors that are likely caused by the Zen Storage.
I solved the first error "Unreal Zen Storage cannot determine a valid path"/"Unreal Zen Storage Server needs to be updated to a new version" by setting the Zen's AutoLaunch to false in home/cs/UnrealEngine/Engine/Config/BaseEngine.ini (and in the other .ini files too, but this did not seem to have an effect).
Now I have a different probably Zen related error: "Fatal error: [File:Developer/DerivedDataCache/Private/DerivedDataBackends.cpp] [Line: 214] 
Unable to use default cache graph 'DerivedDataBackendGraph' because there are no readable or writable nodes available.Add -DDC-ForceMemoryCache to the command line to bypass this if you need access to the editor settings to fix the cache configuration." The full crash report is below.
Next, the Unreal forum suggests to start the editor with the additional argument -ddc=NoZenLocalFallback, but since I use `cmake --build Build/Release/ --target launch` to launch the editor, this did not work. I assume since I want to start the Carla Unreal Editor and not just Unreal, that the solution might look different, that's why I post here and not in the Unreal forum.

I highly appreciate any ideas on how to solve this. Thanks in advance for your help!

**Solutions I tried that did not work:**
- Some recommended to switch off a VPN or to change networks. I do not use a VPN and changing networks did not solve the problem. 
- I deleted the DerivedDataCache folder
- I changed the path for the Zen.AutoLaunch like this: https://forums.unrealengine.com/t/unreal-5-4-crash-will-not-start-unreal-zen-storage-server-failed-to-launch/1765413/13

**Expected behavior**
`cmake --build Build/Release/ --target launch` should launch the editor.

**Logs**
Crash Report of the Unreal Editor:
LoginId:34c4dd10f5c94f5da3f7731615869921-000003e8

Fatal error: [File:Developer/DerivedDataCache/Private/DerivedDataBackends.cpp] [Line: 214] 
Unable to use default cache graph 'DerivedDataBackendGraph' because there are no readable or writable nodes available.Add -DDC-ForceMemoryCache to the command line to bypass this if you need access to the editor settings to fix the cache configuration.
0x00007fa95ea74634 libUnrealEditor-DerivedDataCache.so!UE::DerivedData::FDerivedDataBackendGraph::FDerivedDataBackendGraph() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataBackends.cpp:208]
0x00007fa95ea73259 libUnrealEditor-DerivedDataCache.so!UE::DerivedData::FDerivedDataBackend::Create() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataBackends.cpp:1132]
0x00007fa95eaf364e libUnrealEditor-DerivedDataCache.so!UE::DerivedData::Private::FDerivedDataCache::FDerivedDataCache() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCache.cpp:675]
0x00007fa95eaf32f0 libUnrealEditor-DerivedDataCache.so!UE::DerivedData::Private::CreateCache(FDerivedDataCacheInterface**) [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCache.cpp:1240]
0x00007fa95eb54a3f libUnrealEditor-DerivedDataCache.so!UE::DerivedData::GetCache() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCacheModule.cpp:42]
0x0000000000247782 UnrealEditor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3332]
0x0000000000243119 UnrealEditor!FEngineLoop::PreInit(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4429]
0x0000000000279272 UnrealEditor!GuardedMain(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:49]
0x00007fa96955c4aa libUnrealEditor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/cs/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:323]
0x00007fa95b229d90 libc.so.6!UnknownFunction(0x29d8f)
0x00007fa95b229e40 libc.so.6!__libc_start_main(+0x7f)
0x000000000023e85a UnrealEditor!_start(+0x29)

libUnrealEditor-DerivedDataCache.so!UE::DerivedData::FDerivedDataBackendGraph::FDerivedDataBackendGraph() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataBackends.cpp:208]
libUnrealEditor-DerivedDataCache.so!UE::DerivedData::FDerivedDataBackend::Create() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataBackends.cpp:1132]
libUnrealEditor-DerivedDataCache.so!UE::DerivedData::Private::FDerivedDataCache::FDerivedDataCache() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCache.cpp:675]
libUnrealEditor-DerivedDataCache.so!UE::DerivedData::Private::CreateCache(FDerivedDataCacheInterface**) [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCache.cpp:1240]
libUnrealEditor-DerivedDataCache.so!UE::DerivedData::GetCache() [/home/cs/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/DerivedDataCacheModule.cpp:42]
UnrealEditor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3332]
UnrealEditor!FEngineLoop::PreInit(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4429]
UnrealEditor!GuardedMain(char16_t const*) [/home/cs/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:49]
libUnrealEditor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/cs/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:323]
libc.so.6!UnknownFunction(0x29d8f)
libc.so.6!__libc_start_main(+0x7f)
UnrealEditor!_start(+0x29)

MarcelPiNacy-CVC, 2025-03-19:
Is this a clean install of the engine?

MarcelPiNacy-CVC, 2025-04-03:
I'm closing this, feel free to reopen it later.

closed 2025-04-03 by MarcelPiNacy-CVC