carla-simulator/carla #8751: Build CARLA from Source on Windows problem

Rizkyhamm, 2025-03-13:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: Windoes 10
  * Python version: Python 3.8.5
  * GPU: NVIDIA GeForce RTX 3090
  * GPU Drivers: NVIDIA CUDA

**Describe the bug**
So, i want to Build CARLA from Source based on this docs https://carla.readthedocs.io/en/0.9.15/build_windows/. But, when i build with "make PythonAPI", the result is like this:

D:\Rizky_RA\carla>cmake --version
cmake version 4.0.0-rc3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

D:\Rizky_RA\carla>make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32

D:\Rizky_RA\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "D:\Rizky_RA\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "D:\Rizky_RA\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\Rizky_RA\carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "D:\Rizky_RA\carla\Build\zlib-source\".
CMake Error: The source directory "D:/Rizky_RA/carla/Build/zlib-source"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

**Additional context**
The version of CMake is 4.0.0-rc3 and Make version is GNU Make 3.81.

[image]

rifatrakib, 2025-03-14:
I am also facing the same issue. I am on Windows 11 machine, python 3.12, GPU NVIDIA RTX 4060. One more thing, I am using Visual Studio 2022 as I had it installed previously, and I cannot download the 2019 version from their website currently.

Rizkyhamm, 2025-03-19:
For the 2019 VS version, i get from Microsoft's [website](https://my.visualstudio.com/Downloads?q=visual%20studio%202019&wt.mc_id=o~msft~vscom~older-downloads). And for the issue, i solved by extracting this zlib 1.2.11 below and extracted on \Builds directory.

[zlib1_2_11.zip](https://github.com/user-attachments/files/19333103/zlib1_2_11.zip)

closed 2025-03-19 by Rizkyhamm

namlevitan, 2025-03-23:
For this issue please install VS2019 community 

[image]

As you can see, now the only thing I'm facing is boostrap extraction

[image]

UniAdaRepo, 2025-03-25:
Thank you so much! I had the same issue and solved by using zlib 1.2.11 instead of zlib 1.3. Since in zlib 1.3 some files are missing

Rizkyhamm, 2025-04-11:
Yeah, i'm happy that your issue was solved!