carla-simulator/carla #8771: [0.9.15 / 0.10.0] LIDAR passes through dynamic objects when replaying recording

dmv5383, 2025-03-19:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15 and 0.10.0
  * Platform: Ubuntu 22
  * Python version: 3.8 and 3.10
  * GPU: GTX 1070

**Describe the bug**
The LIDAR simulation is only able to interact with the map and passes straight through dynamic objects such as vehicles and walkers. This seems to be an issue with physics being disabled when replaying a record. Is there any way to enable the physics simulation?

Based on #7092 I tried setting vehicle.set_collisions(True) and vehicle.set_simulate_physics(True) but it didn't work.

I also tried upgrading from 0.9.15 to 0.10.0 but the issue persisted.

**Screenshots**

[image]

dmv5383, 2025-03-19:
Seems to be related to #7092, #7238, and #6911 which say the problem was fixed, but the problem persisted after upgrading.

Blyron, 2025-04-07:
Yeah the issue is in CarlaReplayer.cpp. Line234, I think that line should be         
RootComponent->SetCollisionEnabled(ECollisionEnabled::QueryOnly);

I haven't tested and I did not have time.

glopezdiest, 2025-08-06:
@Blyron @dmv5383 I can confirm that the issue is present in 0.9.15, but current dev has the issue fixed, due to some changes we did to the LIDAR.

As a result, I'm closing this issue. Thank you for pointing it out and feel free to reopen it if there are still unanswered questions.

closed 2025-08-06 by glopezdiest

st3lzer, 2025-08-12:
@glopezdiest do you mean the ue5-dev branch? I want to use record files long term, can I use 0.9.15 record files also with this branch, which is presumably closer to 0.10.0 or do I need to use 0.10.0/ue5-dev to capture the recording files for replaying?

Blyron, 2025-08-19:
They should be