carla-simulator/carla #8774: Huge problems with Carla Dev Branch

Vincent318, 2025-03-20:
**Setup**
Describe the setup you are using to run CARLA along with its version:
CARLA version: 0.9.15
Platform: Unreal Engine 4.26
Python version: 3.8.17
GPU: NVIDIA RTX A2000
GPU Drivers: NVIDA-SMI 553.35

**Describe the bug**
Hello, I built Carla from source many times and have many problems which always occur...cooking and packaging fails, Version mismatch, RGB-Error etc. It always was the dev branch where stands that there might be some unexpected behavior. When I go to the other guides, for example 0.9.14 or 0.9.15 instead of latest, it doesn't function at all because the git clone https://github.com/carla-simulator/carla refers to CarlaUE5 which is wrong. Can please someone give me a guide which functions without these problems? Thank you for your help! I really need it for my Bachelor Thesis.

**Steps to reproduce**
Build Carla from Source like shown on https://carla.readthedocs.io/en/0.9.15/build_windows/

**Expected behavior**
It should work without problems

**Logs**

**Scripts**

**Screenshots**

**Additional context**

Blyron, 2025-03-21:
Please use the binaries provided or use this docs;
https://carla.readthedocs.io/en/latest/build_windows/
If you want to download specific version git is your friend, there are tags pointing each version 
https://git-scm.com/book/en/v2/Git-Basics-Tagging
Also, the update.bat for content is not work for you. 
You will need to clone manually the content into 

Unreal/CarlaUE4/Content/ 

The content repo link with the tag;
https://bitbucket.org/carla-simulator/carla-content/src/0.9.15/

Vincent318, 2025-03-24:
Hello,
Thanks for your answer. I tried to build it this way, but unfortunately, the build failes.
I followed the build guide https://carla.readthedocs.io/en/latest/build_windows/ and after the Carla Clone, I made git checkout tags/0.9.15 -b my-0.9.15-branch and there was shown that I'm on tag 0.9.15. I cloned the maps with git clone --branch 0.9.15 https://bitbucket.org/carla-simulator/carla-content.git Unreal/CarlaUE4/Content/, this all functions. But when I try to make PythonAPI after that, there is the error:

Ausnahme beim Aufrufen von "DownloadFile" mit 2 Argument(en):  "Der Remoteserver hat einen Fehler zurückgegeben: (404)  Nicht gefunden."                                                                                                        In Zeile:1 Zeichen:1                                                                                                    + (New-Object System.Net.WebClient).DownloadFile('https://www.zlib.net/ ...                                             + ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                     + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException

    + FullyQualifiedErrorId : WebException

    -[install_zlib]: Retrieving zlib from backup.

Ausnahme beim Aufrufen von "DownloadFile" mit 2 Argument(en):  "Der Remoteserver hat einen Fehler zurückgegeben: (403)

Unzulässig."

In Zeile:1 Zeichen:1

+ (New-Object System.Net.WebClient).DownloadFile('https://carla-release ...

+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException

    + FullyQualifiedErrorId : WebException

    -[install_zlib]: Extracting zlib from "zlib-1.2.13.zip".

Expand-Archive : Der Pfad "D:\carla\carla\Build\zlib-1.2.13.zip" ist entweder nicht vorhanden oder entspricht keinem

gültigen Dateisystempfad.

In Zeile:1 Zeichen:1

+ Expand-Archive 'D:\carla\carla\Build\zlib-1.2.13.zip' -DestinationPat ...

+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

    + CategoryInfo          : InvalidArgument: (D:\carla\carla\Build\zlib-1.2.13.zip:String) [Expand-Archive], Invalid

   OperationException

    + FullyQualifiedErrorId : ArchiveCmdletPathNotFound,Expand-Archive

    -[install_zlib]: Removing "zlib-1.2.13.zip"

D:\carla\carla\Build\zlib-1.2.13.zip konnte nicht gefunden werden

Das System kann die angegebene Datei nicht finden.

    -[install_zlib]: Creating "D:\carla\carla\Build\zlib-source\build"

CMake Error: The source directory "D:/carla/carla/Build/zlib-source" does not appear to contain CMakeLists.txt.

Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.

    -[install_zlib]: [CMAKE ERROR] Possible causes:

    -[install_zlib]:                - Make sure "CMake" is installed.

    -[install_zlib]:                - Make sure it is available on your Windows "path".

    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.

    -[install_zlib]: Exiting with error...

-[Setup]:

   Ok, and error ocurred, don't panic

   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:

       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:

       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel

       We'll be glad to help you there :)

       https://discord.gg/42KJdRj

make: *** [setup] Fehler 1

Do you or somebody other know why this error occurs and what I should make to solve this? Maybe there's something which I didn't made right, for example the git tag commands?

Blyron, 2025-04-01:
I agree with you that we need more robust build system and docs. But there are too many different setups configs which we cannot test. That's why we are open to PRs. 
Our mayority of users are professionals and usually know how to deal with this issues.

qhaas, 2025-06-08:
I followed this build guide:  https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/#extended-build-instructions

Unfortunately, the 0.10.0 tags did not build, so I used the commits noted here: https://github.com/carla-simulator/carla/issues/8892#issuecomment-2935687659