carla-simulator/carla #8780: Make CARLA cache folder configurable

Kait0, 2025-03-23:
**Is your feature request related to a problem? Please describe.**
I am trying to run many CARLA instances on a cluster of servers which uses a slow shared a file system.
Each node has a copy of the CARLA source files on its local storage. However, since the OS uses a shared file system, and $HOME is on the shared file system, all my CARLA copies use the same cache folder in the slow shared file system.

**Describe the solution you'd like**
Make the cacheFolder configurable by first checking if an environment variable is set (say CARLA_CACHE_FOLDER).
If it is set, select the CARLA_CACHE_FOLDER destination, else use the default ~/carlaCache option.
Should be rather easy to implement and is backwards compatible.

**Describe alternatives you've considered**
The standard solution to symlink the Cache folder doesn't work here, since every node needs a different destination for the symlink (since they can't access each other's local storage).

Kait0, 2025-03-27:
Thank you for the pointer. It is good to know that the base folder can be changed after starting the server.
Unfortunately, this doesn't quite solve my use case. What I am trying to do is start >100 CARLA servers simultaneously, which on shared file systems seems to hang up (but works on local storage). So I would like to change the cache folder before connecting a client.

Blyron, 2025-04-03:
You are right. We will do this.

Blyron, 2025-04-03:
This is now working on ue5-dev, we are doing in ue4-dev soon

closed 2025-04-04 by Blyron