carla-simulator/carla #8783: Unable to succesfully build Carla in Windows

LauraGBodego, 2025-03-24:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: Windows 11 and Windows 10
  * Python version: 3.8.0
  * GPU: NVIDIA GeGorde RTX 4090 Laptop GPU
  * GPU Drivers: 32.0.15.7270

**Describe the bug**
I am trying to compile CARLA 0.9.15 on Windows using Visual Studio 2022. Despite following the official Windows build tutorial and multiple GitHub Issues and blog posts, the compilation fails with errors related to missing dependencies and issues with STL headers. The most critical errors I encountered are:

Compilation errors in OutputDevice_File.cpp, stating that std::basic_ofstream<char,std::char_traits<char>> is undefined, even though <fstream> is included.

Missing directory error: FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib' while building Python API.

**Steps to reproduce**
1. Clone Unreal Engine for CARLA:
git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git

2. Run setup.bat (Replaced Commit.gitdeps.xml due to errors).

3. Generate project files and build Unreal Engine.

4. Clone CARLA repository:
git clone https://github.com/carla-simulator/carla -b 0.9.15

5. Set UE4_ROOT environment variable.

6. Modify batch scripts to reference Visual Studio 2022 instead of 2019.

7. Run update.bat (Downloaded assets manually after failure).

8. Modify install_zlib.bat to use GitHub links instead of broken zlib.net.

9. Run make PythonAPI and encounter the missing dependencies/lib directory error.

**Attempts to Fix**

* Tried compiling with Visual Studio 2019, but encountered similar errors.

* Updated all batch scripts to reference s3.us-east-005.backblazeb2.com instead of s3.amazonaws.com for dependencies.

* Manually downloaded missing .tar.gz assets.
* Changed buildPytonAPI.bat line 112
    py -3 setup.py bdist_egg bdist_wheel ->   "C:\Program Files\Python38\python.exe" setup.py bdist_egg bdist_wheel
* Manually downloaded and unZip in Build\ folder :
   boost-1.80.0-install and boost-1.80.0-source

**Logs**

C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Exception.cpp(8,10): error C1083: No se puede abrir el archivo incluir: 'boost/assert/source_location.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\StringUtil.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/algorithm/string.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/StringUtil.cpp")

  Walker.cpp
  WalkerAIController.cpp
C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/client/Light.cpp")

C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/client/Junction.cpp")

  World.cpp
  ActorFactory.cpp
C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/client/World.cpp")

  ActorState.cpp
C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/client/detail/ActorFactory.cpp")
[... 422 lines cut ...]
      C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.43.34808\include\iosfwd(207,7):
      vea la declaraci├│n de 'std::basic_ofstream<char,std::char_traits<char>>'

El sistema no puede encontrar la ruta especificada.
El sistema no puede encontrar la ruta especificada.
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Proyectos\CarlaBuild\carla-0.9.15\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 169, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 122, in get_libcarla_extensions
    libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
FileNotFoundError: [WinError 3] El sistema no puede encontrar la ruta especificada: 'dependencies/lib'

**Additional context**
Compilation has been attempted with Visual Studio 2022 and Visual Studio 2019: Same errors, although setup.bat worked without needing to modify Commit.gitdeps.xml.

Other users' guides and solutions have also been followed without success:

[Issue #6436](https://github.com/carla-simulator/carla/issues/6436) (dependencies/lib error)

[Issue #6898](https://github.com/carla-simulator/carla/issues/6898) (BuildOSM2ODR error)

[Issue #2642](https://github.com/carla-simulator/carla/issues/2642) (setup.bat error, although in my case, CMakeLists.txt seems correct).

Blyron, 2025-03-27:
The 1st step is wrong. It is clonning ue5-dev-carla branch instead of carla branch. 
Try to switch to "carla" branch in UE fork repo

Daniel5888tw, 2025-04-01:
@LauraGBodego I got the same error as you. Do you have any solution?

LauraGBodego, 2025-04-01:
> [@LauraGBodego](https://github.com/LauraGBodego) I got the same error as you. Do you have any solution?

Not yet :(

MarcelPiNacy-CVC, 2025-04-03:
Hi, try deleting the boost folders inside build, running make setup and then make LibCarla. Share the output logs from that.

LauraGBodego, 2025-04-04:
> Hi, try deleting the boost folders inside build, running make setup and then make LibCarla. Share the output logs from that.

It still doesn't work

Logs from deleting boost and running make setup:

C:\Proyectos\CarlaBuild\carla-0.9.15>make setup
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Proyectos\CarlaBuild\carla-0.9.15\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 17 2022"
-[Setup]: Install directory:  "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\" --zlib-install-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\" --generator "Visual Studio 17 2022"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\" --generator "Visual Studio 17 2022"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\" --generator "Visual Studio 17 2022"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Proyectos\CarlaBuild\carla-0.9.15\Build\" --toolset msvc-14.2 --version 1.80.0 -j 20
    -[install_boost]: Retrieving boost.
    -[install_boost]: Extracting boost from "boost_1_80_0.zip", this can take a while...

7-Zip 24.09 (x64) : Copyright (c) 1999-2024 Igor Pavlov : 2024-11-29

Scanning the drive for archives:
1 file, 11534 bytes (12 KiB)

Extracting archive: C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost_1_80_0.zip
ERROR: C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost_1_80_0.zip
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost_1_80_0.zip
Open ERROR: Cannot open the file as [zip] archive

ERRORS:
Is not archive

Can't open as archive: 1
Files: 0
Size:       0
Compressed: 0
    -[install_boost]: Removing "boost_1_80_0.zip"
El sistema no puede encontrar el archivo especificado.
El sistema no puede encontrar la ruta especificada.
    -[install_boost]: Generating build...
"bootstrap.bat" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.

[... 1447 lines cut ...]
C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/trafficmanager/Parameters.cpp")

C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/trafficmanager/TrafficLightStage.cpp")

C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Memory.h(9,10): error C1083: No se puede abrir el archivo incluir: 'boost/enable_shared_from_this.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp")

C:\Proyectos\CarlaBuild\carla-0.9.15\LibCarla\source\carla\Buffer.h(13,10): error C1083: No se puede abrir el archivo incluir: 'boost/asio/buffer.hpp': No such file or directory [C:\Proyectos\CarlaBuild\carla-0.9.15\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/ros2/ROS2.cpp")

[... cut ...]
  (compilando archivo de origen "../../../../../LibCarla/source/carla/trafficmanager/TrafficManager.cpp")

-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\Proyectos\CarlaBuild\carla-0.9.15\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "C:\Proyectos\CarlaBuild\carla-0.9.15\PythonAPI\carla\dependencies\"!

I've checked,  "Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\" and "PythonAPI\carla\dependencies\" have not been created

LauraGBodego, 2025-04-04:
> The issue is boost not being downloaded Could you share the line 75 of install_boost.bat in Util/InstallersWin?

Line 75 install_boost.bat:

set BOOST_SRC_DIR=%BUILD_DIR%%BOOST_BASENAME%-source\

Blyron, 2025-04-04:
which url is trying to get? you should be able to see an url in the same file. 
Btw,  have you tried ue4-dev branch?

LauraGBodego, 2025-04-04:
> which url is trying to get? you should be able to see an url in the same file. Btw, have you tried ue4-dev branch?

Do you mean this url?

Blyron, 2025-04-04:
En el archivo aparece un link, AKA, url. Que es donde esta descargando la version de boost. Tiene toda la pinta de que el link al que apunta el archvio en la version 0.9.15.2 esta roto. Asi que, habria que cambiar ese link.

I mean this one;
set BOOST_REPO=https://archives.boost.io/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
Is that the same for you?

LauraGBodego, 2025-04-04:
> En el archivo aparece un link, AKA, url. Que es donde esta descargando la version de boost. Tiene toda la pinta de que el link al que apunta el archvio en la version 0.9.15.2 esta roto. Asi que, [...]

He cambiado el link porque era diferente en mi boost.bat, 
(en el mío era 
set BOOST_REPO=https://boostorg.jfrog.io/artifactory/main/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE% y lo he cambiado por 
set BOOST_REPO=https://archives.boost.io/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%)

parecía que funcionaba pero al final me ha dado error en el b2.exe

C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\cast.hpp
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\cerrno.hpp
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\cerrno.hpp
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\checked_delete.hpp
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\checked_delete.hpp
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\chrono.hpp
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\chrono.hpp
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cassert
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cassert
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cctype
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cctype
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cerrno
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cerrno
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cfloat
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cfloat
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\climits
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\climits
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\clocale
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\clocale
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cmath
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cmath
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\csetjmp
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\csetjmp
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\csignal
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\csignal
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cstdarg
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cstdarg
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cstddef
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cstddef
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cstdio
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cstdio
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cstdlib
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cstdlib
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cstring
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cstring
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\ctime
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\ctime
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cwchar
[... cut ...] pp_c_headers\cwchar
        1 archivo(s) copiado(s).
common.copy C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-install\boost-1_80\boost\compatibility\cpp_c_headers\cwctype
C:\Proyectos\CarlaBuild\carla-0.9.15\Build\boost-1.80.0-source\boost\compatibility\cpp_c_headers\cwctype
        1 archivo(s) copiado(s).
...skipped 63 targets...
...updated 16474 targets...

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...

MarcelPiNacy-CVC, 2025-04-04:
Podrias enviar todo el log? Con `make setup >> Log.txt`.

Blyron, 2025-04-04:
Mi consejo esque utlices la ue4-dev branch.
Ademas si lo estas haciendo con Visual Studio  2022; Neceistaras esto;

LauraGBodego, 2025-04-04:
> Podrias enviar todo el log? Con `make setup >> Log.txt`.

[Log.txt](https://github.com/user-attachments/files/19602107/Log.txt)

LauraGBodego, 2025-04-04:
> Mi consejo esque utlices la ue4-dev branch. Ademas si lo estas haciendo con Visual Studio 

> Mi consejo esque utlices la ue4-dev branch. Ademas si lo estas haciendo con Visual Studio 2022; Neceistaras esto; https://docs.google.com/document/d/1JmHQiLii-wehCe0JyHSpvRiI54i9IaxPLOgWSeHb6Mk/edit?tab=t.0

Gracias, lo  intento de esta forma

MarcelPiNacy-CVC, 2025-04-04:
Si, yo estaba teniendo este mismo error. Los cambios que comenta Blyron arreglan este problema.

LauraGBodego, 2025-04-07:
> Si, yo estaba teniendo este mismo error. Los cambios que comenta Blyron arreglan este problema.

> Mi consejo esque utlices la ue4-dev branch. Ademas si lo estas haciendo con Visual Studio 2022; Neceistaras esto; https://docs.google.com/document/d/1JmHQiLii-wehCe0JyHSpvRiI54i9IaxPLOgWSeHb6Mk/edit?tab=t.0

Lo he estado intentando hacerlo desde la rama de ue4-branch pero me sigue dando error en el b2.exe:

[log.txt](https://github.com/user-attachments/files/19628324/log.txt)

Blyron, 2025-04-07:
Pues hay algo en tu sistema que no va bien. Borralo todo clonalo de nuevo y sigue la doc paso por paso;
Usa Python 3.10. 
https://carla.readthedocs.io/en/latest/build_windows/

Esto tiene que funcionar y esta testeado

LauraGBodego, 2025-04-08:
> Pues hay algo en tu sistema que no va bien. Borralo todo clonalo de nuevo y sigue la doc paso por paso; https://carla.readthedocs.io/en/latest/build_windows/
> 
> Esto tiene que funcionar y esta testeado

Ya he conseguido compilarlo, pero al hacer el siguiente paso (make launch) me sale el siguiente error:
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Proyectos\CARLA\carla\PythonAPI\carla\dependencies\"
-[BuildCarlaUE4]: [Batch params]: --build --all
--build
--all
ECHO está desactivado.
-[BuildCarlaUE4]: Building Unreal Editor...
Using bundled DotNet SDK version: 8.0.300
Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" CarlaUE4Editor        Win64        Development        -WaitMutex        -FromMsBuild        "C:/Proyectos/CARLA/carla/Unreal/CarlaUE4/CarlaUE4.uproject"
Log file: C:\Users\1M06330\AppData\Local\UnrealBuildTool\Log.txt
Creating makefile for CarlaUE4Editor (no existing makefile)
Compiled assembly file 'C:\Proyectos\CARLA\carla\Unreal\CarlaUE4\Intermediate\Build\BuildRules\CarlaUE4ModuleRules.dll' appears to be for a newer CLR version or is otherwise invalid.  Unreal Build Tool will try to recompile this assembly now.  (Exception: Bad IL format. The format of the file 'C:\Proyectos\CARLA\carla\Unreal\CarlaUE4\Intermediate\Build\BuildRules\CarlaUE4ModuleRules.dll' is invalid.)
C:\Proyectos\CARLA\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Carla.Build.cs(15,104): error CS0117: 'UnrealTargetPlatform' does not contain a definition for 'Win32'
Total execution time: 20.34 seconds
Expecting to find a type to be declared in a target rules named 'CarlaUE4EditorTarget'.  This type must derive from the 'TargetRules' type defined by UnrealBuildTool.
make: *** [CarlaUE4Editor] Error 8

C:\Proyectos\CARLA\carla>

Blyron, 2025-04-08:
Your machine is x64 or x32?

LauraGBodego, 2025-04-09:
> Your machine is x64 or x32?

X64

Blyron, 2025-04-09:
Que version de Python usas?

Blyron, 2025-04-09:
Borra la carpeta Binaries e Intermediate en 
Unreal\CarlaUE4 y Unreal\CarlaUE4\Plugins\Carla 
a ver si eso si. 
No hemos visto ese error nunca

LauraGBodego, 2025-04-09:
> Borra la carpeta Binaries e Intermediate en Unreal\CarlaUE4 y Unreal\CarlaUE4\Plugins\Carla a ver si eso si. No hemos visto ese error nunca

Creo que es porque la ruta de unreal en las variables de sistema la tenía puesta como UE4-ROOT en lugar de UE4_ROOT, aunque después de ese cambio me ha dado otro error diferente:

-- Build files have been written to: C:/Proyectos/CARLA/carla/Build/osm2odr-visualstudio
Versión de MSBuild 17.13.15+18b3035f6 para .NET Framework

  osm2odr.vcxproj -> C:\Proyectos\CARLA\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  1>
  -- Install configuration: "Release"
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Proyectos\CARLA\carla\PythonAPI\carla\dependencies\"
-[BuildCarlaUE4]: [Batch params]: --build --all
--build
--all
ECHO está desactivado.
-[BuildCarlaUE4]: Building Unreal Editor...
El sistema no puede encontrar la ruta especificada.
make: *** [CarlaUE4Editor] Error 1

Blyron, 2025-04-09:
Ahora que tienes la variable bien seteada mi consejo es que hagas un 
git clean -fdx en el carla root y hagas de nuevo un make PythonAPI y make Launch

LauraGBodego, 2025-04-09:
> Ahora que tienes la variable bien seteada mi consejo es que hagas un git clean -fdx en el carla root y hagas de nuevo un make PythonAPI y make Launch

Al hacer eso me ha vuelto a dar error make python api

[log.txt](https://github.com/user-attachments/files/19666361/log.txt)

Blyron, 2025-04-09:
No te esta compilando el libproj. Prueba a bajarte manualmente de aqui la dependencia manualmente 
https://download.osgeo.org/proj/
la version 7.2.1
y lo tiene que poner en la carpeta build nombrado como "proj-src"
La carpeta Build tiene que quedar asi 

[image]

LauraGBodego, 2025-04-10:
> No te esta compilando el libproj. Prueba a bajarte manualmente de aqui la dependencia manualmente https://download.osgeo.org/proj/ la version 7.2.1 y lo tiene que poner en la carpeta build nombrado como "proj-src" La carpeta Build tiene que quedar asi
> 
> [image]

Gracias! he hecho esos cambios y funciona make pythonApi, pero me sigue dando el mismo error de antes en make launch

C:\Proyectos\CARLA\carla>make launch
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Proyectos\CARLA\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.3 --generator "" --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.3
-[Setup]: Generator:          "Visual Studio 17 2022"
-[Setup]: Install directory:  "C:\Proyectos\CARLA\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Proyectos\CARLA\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --zlib-install-dir "C:\Proyectos\CARLA\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Proyectos\CARLA\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --generator "Visual Studio 17 2022"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Proyectos\CARLA\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --generator "Visual Studio 17 2022"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Proyectos\CARLA\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --generator "Visual Studio 17 2022"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Proyectos\CARLA\carla\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --toolset msvc-14.3 --version 1.80.0 -j 20
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Proyectos\CARLA\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --generator "Visual Studio 17 2022"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Proyectos\CARLA\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\Proyectos\CARLA\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\" --generator "Visual Studio 17 2022"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\Proyectos\CARLA\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\Proyectos\CARLA\carla\Build\"
0 archivo(s) copiado(s)
[... 127 lines cut ...]
--     CMake build tool: C:/Program Files/Microsoft Visual Studio/2022/Professional/MSBuild/Current/Bin/amd64/MSBuild.exe
--     Compiler: MSVC 19.36.32548.0
--
-- Found Proj: C:/Proyectos/CARLA/carla/Build/proj-install/lib/proj.lib
-- Enabled features: Windows-10.0.26100 AMD64 MSVC 19.36.32548.0 Release Proj
-- Configuring done (0.4s)
-- Generating done (0.6s)
-- Build files have been written to: C:/Proyectos/CARLA/carla/Build/osm2odr-visualstudio
Versión de MSBuild 17.13.15+18b3035f6 para .NET Framework

  osm2odr.vcxproj -> C:\Proyectos\CARLA\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  1>
  -- Install configuration: "Release"
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Proyectos\CARLA\carla\PythonAPI\carla\dependencies\"
-[BuildCarlaUE4]: [Batch params]: --build --all
--build
--all
ECHO está desactivado.
-[BuildCarlaUE4]: Building Unreal Editor...
El sistema no puede encontrar la ruta especificada.
make: *** [CarlaUE4Editor] Error 1

C:\Proyectos\CARLA\carla>

Blyron, 2025-04-10:
Como luce ahora tu carpeta Build?

LauraGBodego, 2025-04-10:
> Como luce ahora tu carpeta Build?

[image]

MarcelPiNacy-CVC, 2025-04-15:
Run this: `git clean -fxd && git reset --hard && git pull`
Then run this command `make setup >> Setup.log` and send here Setup.log.

LauraGBodego, 2025-04-16:
> Run this: `git clean -fxd && git reset --hard && git pull` Then run `make setup >> Setup.log` and send here Setup.log.

[image]

[image]

[image]

[setup.log](https://github.com/user-attachments/files/19773587/setup.log)

MarcelPiNacy-CVC, 2025-04-16:
Ok, now run `make LibCarla >> LibCarla.log`, send LibCarla.log here too.

LauraGBodego, 2025-04-16:
> Ok, now run `make LibCarla >> LibCarla.log`, send LibCarla.log here too.

[libcarla.log](https://github.com/user-attachments/files/19774512/libcarla.log)

MarcelPiNacy-CVC, 2025-04-16:
Looks good! Now run `make PythonAPI >> PythonAPI.log`. Send PythonAPI.log here too.

LauraGBodego, 2025-04-16:
> Looks good! Now run `make PythonAPI >> PythonAPI.log`. Send PythonAPI.log here too.

[image]

[PythonAPI.log](https://github.com/user-attachments/files/19778872/PythonAPI.log)

MarcelPiNacy-CVC, 2025-04-16:
Could you share a screenshot of the folder `c:\Proyectos\CARLA\carla\PythonAPI\carla\dependencies\`?

MarcelPiNacy-CVC, 2025-04-16:
Also one of the CARLA Build folder (That is: `c:\Proyectos\CARLA\carla\Build`)

LauraGBodego, 2025-04-16:
> Also one of the CARLA Build folder (That is: `c:\Proyectos\CARLA\carla\Build`)

Sure!

[image]

[image]

LauraGBodego, 2025-04-21:
> Try deleting the osm2odr-visualstudio folder in Build and retry `make PythonAPI`.

[pythonAPI.log](https://github.com/user-attachments/files/19830970/pythonAPI.log)

MarcelPiNacy-CVC, 2025-04-25:
Do you have UE4_ROOT set as an environment variable?

LauraGBodego, 2025-04-25:
> Do you have UE4_ROOT set as an environment variable?

Yes 

[image]

MarcelPiNacy-CVC, 2025-04-25:
UE4_ROOT should point at the engine root, not its build folder. That's C:\Proyectos\UnrealEngine

LauraGBodego, 2025-04-25:
> UE4_ROOT should point at the engine root, not its build folder. That's C:\Proyectos\UnrealEngine

Ok, I'll run it again with that changed

MarcelPiNacy-CVC, 2025-04-28:
How did it go?

LauraGBodego, 2025-05-06:
> How did it go?

I've just tried it, and it seemed to be working but it gives a fatal error while opening unreal 

[image]

MarcelPiNacy-CVC, 2025-05-06:
Share the entire log from unreal. It will be in your Saved folder.

LauraGBodego, 2025-05-07:
> Share the entire log from unreal. It will be in your Saved folder.

This one?

[CarlaUE4.log](https://github.com/user-attachments/files/20077585/CarlaUE4.log)

MarcelPiNacy-CVC, 2025-05-07:
You don't seem to have content downloaded, run Update.bat.

LauraGBodego, 2025-05-08:
> You don't seem to have content downloaded, run Update.bat.

Thank you, that was it.
After that i got another error:

[image]

But i fixed it changing the permissions in the unreal build folder and rebuilding Unreal

Now is working, thank you so much for all your help :)

MarcelPiNacy-CVC, 2025-05-08:
Wonderful! I'm closing this issue for now then, feel free to reopen it if needed.

closed 2025-05-08 by MarcelPiNacy-CVC