carla-simulator/carla #8799: Carla ue5-dev instsall failed in windows 11

dedoogong, 2025-03-29:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: latest ( ue5-dev)
  * Platform: Win11 
  * Python version: 3.10
  * GPU: 4090
  * GPU Drivers: 556.12 / CUDA 12.6
  * 
**Describe the bug**
install stopped after some successful steps with a strange error
✅ Installing prerequisites...
✅ Found Ninja.
✅ Found Python.
...
✅ git clone Carla unreal engine 
✅ build Carla unreal engine 
✅ CARLA Python API build+install succeeded.
✅ Launching Carla Unreal Editor...
✅ Found CARLA content.
....

❌ [0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

**Steps to reproduce**
CarlaSetup.bat --python-root=C:\Users\USER\AppData\Local\Programs\Python\Python310

**Expected behavior**
all related modules are installed correctly

**Logs**
CarlaSetup.bat --python-root=C:\Users\USER\AppData\Local\Programs\Python\Python310
python_path is: python
python_path is: C:\Users\USER\AppData\Local\Programs\Python\Python310\python
Installing prerequisites...
Found Ninja.
Found Python.
Requirement already satisfied: pip in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (25.0.1)
Requirement already satisfied: psutil in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 1)) (7.0.0)
Requirement already satisfied: requests in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 2)) (2.32.3)
Requirement already satisfied: scikit-build-core in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 3)) (0.11.1)
Requirement already satisfied: wheel in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 4)) (0.45.1)
Requirement already satisfied: numpy<2.0,>=1.24.4 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 5)) (1.26.4)
Requirement already satisfied: setuptools>=47.3.1 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 6)) (65.5.0)
Requirement already satisfied: build in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 7)) (1.2.2.post1)
Requirement already satisfied: pygame in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from -r requirements.txt (line 8)) (2.6.1)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from requests->-r requirements.txt (line 2)) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from requests->-r requirements.txt (line 2)) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from requests->-r requirements.txt (line 2)) (2.3.0)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from requests->-r requirements.txt (line 2)) (2025.1.31)
Requirement already satisfied: exceptiongroup>=1.0 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from scikit-build-core->-r requirements.txt (line 3)) (1.2.2)
Requirement already satisfied: packaging>=23.2 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from scikit-build-core->-r requirements.txt (line 3)) (24.2)
Requirement already satisfied: pathspec>=0.10.1 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from scikit-build-core->-r requirements.txt (line 3)) (0.12.1)
Requirement already satisfied: tomli>=1.2.2 in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from scikit-build-core->-r requirements.txt (line 3)) (2.2.1)
Requirement already satisfied: pyproject_hooks in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from build->-r requirements.txt (line 7)) (1.2.0)
Requirement already satisfied: colorama in c:\users\user\appdata\local\programs\python\python310\lib\site-packages (from build->-r requirements.txt (line 7)) (0.4.6)
Found CARLA content.
Activating "x64 Native Tools Command Prompt" terminal environment.
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.13.5
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'
Found CARLA Unreal Engine at C:\Users\USER\Documents\CarlaUE5/UnrealEngine5_carla. Assuming already built...
Configuring the CARLA CMake project...
-- CARLA: Using  as Unreal Engine root path.
-- CARLA: Checking for /EHsc support
-- CARLA: Checking for /GR support
-- CARLA: Downloading sqlite3...
-- CARLA: Downloading zlib...
-- CARLA: Downloading libpng...
[... 110 lines cut ...]
Building wheels for collected packages: carla
  Building editable for carla (pyproject.toml) ... done
  Created wheel for carla: filename=carla-0.10.0-cp310-cp310-win_amd64.whl size=1462056 sha256=c3bee44587d1eec592c028fed204841154de3a2e48533d356b1173979aaafca2
  Stored in directory: C:\Users\USER\AppData\Local\Temp\pip-ephem-wheel-cache-w8wqhmc8\wheels\7a\30\2a\e5eb7f72c5b5ad7dee6cc7ac87d7d87f0b6912b97228327f81
Successfully built carla
Installing collected packages: carla
  Attempting uninstall: carla
    Found existing installation: carla 0.10.0
    Uninstalling carla-0.10.0:
      Successfully uninstalled carla-0.10.0
Successfully installed carla-0.10.0

CARLA Python API build+install succeeded.
Launching Carla Unreal Editor...
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

**Scripts**
CarlaSetup.bat --python-root=C:\Users\USER\AppData\Local\Programs\Python\Python310

**Screenshots** 

[image]

**Additional context**
Add any other context about the problem here.

MarcelPiNacy-CVC, 2025-04-03:
Try deleting the Build folder and using the manual build steps for now (that is, no CarlaSetup.bat, only CMake).

dedoogong, 2025-04-13:
after reinstalling window 11, MSVC, and all prerequisites, I could finally run carla-ue5. thank you~

closed 2025-04-13 by dedoogong