carla-simulator/carla #8808: Import small size vehicle but failed

HMX2013, 2025-04-02:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: ubuntu 22.04
  * Python version: 3.10
  * GPU: 12th Gen Intel® Core™ i7-12700H × 20
  * GPU Drivers: GeForce RTX 3060 Mobile / Max-Q

**Describe the bug**
I've been trying to import my custom vehicle for two weeks, but I found that it can not run normally in carla-ue4/ue5. The vehicle will tilt forward.

The wheelbase of my vehicle is 0.505 m. The wheel's radius is 10.5cm and 10.5cm in width.

[image]

First, when I import fbx proceesed by blender into ue4, I can only see the vehicle base body(Root). Normally, I should be able to see the other 4 wheel's body. I can only re-generate them manually.

[image]

When I follow the tutorial to complete the rest of the settings and run Carla ue4. The vehicle can not run normally, It will move and tilt forward. Besides, the wheels' position is not correct, they sink a lot.

[image]

I have tried other normal-size vehicles, and they work normally. I have also let other people try it, they also met the same problem.

So could anyone tell me how to solve this problem, I would be quite appreciative. This problem is quite important for my PHD project.

**Steps to reproduce**
Explain the steps needed to reproduce the issue.
The following link includes all your needed files, the blender file, the output fbx file, and the UE4 car import result. You can download and test it.

https://drive.google.com/file/d/1fkQR5TVSmUxXHqrZzgyWja0hFYp6WUnu/view?usp=sharing

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Paste the server and client logs. Server ones can be found at:
  * Unreal Engine 4: _Unreal/CarlaUE4/Saved/Logs_
  * Unreal Engine 5: _Unreal/CarlaUnreal/Saved/Logs_
 no logs in Unreal/CarlaUE4/Saved/Logs_

**Scripts**
Add any scripts relevant for the reproducibility of the error.
https://drive.google.com/file/d/1fkQR5TVSmUxXHqrZzgyWja0hFYp6WUnu/view?usp=sharing

**Screenshots**

[image]

[image]

**Additional context**
Add any other context about the problem here.

gsmario, 2025-04-03:
Hello! As I've seen, your armature is in the wrong position. As the documentation says, it needs to be in the origin. 

[image]

Also, the names of your bones/armature are wrong, you should name them the same as the documentation says to avoid problems.

[image]

Here is the link to the documentation: https://carla.readthedocs.io/en/latest/tuto_content_authoring_vehicles/

Let me know if you have more problems.

HMX2013, 2025-04-03:
https://github.com/carla-simulator/carla/issues/3916

HMX2013, 2025-04-04:
> Hello! As I've seen, your armature is in the wrong position. As the documentation says, it needs to be in the origin.
>
> [image]
>
> Also, the names of your bones/armature are wrong, you should name them [...]

Thank you for your advice. I set the armature to the origin. But it still tilts forward. Nothing changed.

[image]
[image]

You can check the following files if necessary.
https://drive.google.com/file/d/1GnOULhS_c7ZGK9mfAFAi9S4gOPAWujfh/view?usp=sharing

I also tried other modes. I chose a normal-sized car model and then imported it into Carla following the same process. 
It worked perfectly. I can control it manually using manual_control_control.py, located in the PythonAPI folder.
[image]

I tried to shrink the vehicle model a little (wheel radius 20cm, wheel base: 1.42m).  Then import it into Carla. It also worked normally.
https://drive.google.com/file/d/1CPIW84u222wgHsOsUR5bdW2m36kbryef/view?usp=sharing
[image]

But when I further shrink this vehicle (wheel radius 10.5cm, wheel base: 0.75m)). Close to the small-sized car I mentioned above.
It also tilts forward. 

[image]

Should I do some extra settings for a small-sized car? I don't know what I was missing.

gsmario, 2025-04-04:
Hello! Try having a look at the wheel blueprints, there is a static mesh called wheel shape, try to replicate that mesh but with an approximated size of your vehicle. And then test it.
BTW can you add me to discord so we can talk? My user is Kapitastic

HMX2013, 2025-04-05:
> Hello! Try having a look at the wheel blueprints, there is a static mesh called wheel shape, try to replicate that mesh but with an approximated size of your vehicle. And then test it. BTW can you add me to discord so we can talk? My user is Kapitastic

Thank you. It's still not working😂. I have tried to add you in Discord.

gsmario, 2025-04-07:
I've managed to fix the problem creating a new and smaller wheelshape and replacing it in the wheel blueprints

[image]

Then you'll have to adjust the center of mass, since the vehicle is way smaller, the physic values must be more precise than in bigger vehicles.

First drag your vehicle blueprint in to the scene:
You can find the visual center of mass here, (first you need to activate it in the viewport)

[image]

[image]

[image]

You can adjust the values and position here, and once you're happy with the position, you can copy the values, then, go to the real blueprint (open it via content browser) and paste those values the same way. 

As I said, a small variation can change how your vehicle behaves by a lot, so try to adjust it little by little.

Let me know if you need more help!

HMX2013, 2025-04-10:
> I've managed to fix the problem creating a new and smaller wheelshape and replacing it in the wheel blueprints
-----------------------------------------------------------------------------------------------------------------------------

Hello, I have repeated your work. But I still have some problems. 
I found that the rear of the vehicle is moving violently in the vertical direction. especially when I accelerate and decelerate as shown in the following gif.
The center of mass in this case is (0, 0, -13)cm
[image]

I can improve this by changing the center of mass. When I set the center of mass to (10, 0, 14)cm. It's better. But I can still notice the vertical vibration.

[image]

Because I want to install a depth camera on this small-sized vehicle. I don't want the car to vibrate vertically (as small as possible). 
So, how to minimize the vibration of the vehicle in the vertical direction? I know that tuning the center of mass is a way. But it seems that I can not achieve the ideal result just using this. I try to tune other parameters, but I can not get a good result. 
Do you have any suggestions? Are there some parameters I should be careful with? Thank you very much!

LeonTimmer99, 2025-04-14:
@HMX2013 Have you tried changing the suspension parameters in the wheel blueprint?

closed 2025-04-29 by HMX2013

frcs7, 2025-07-28:
@HMX2013 @gsmario I wonder how to deal with wheel body problem，I can only see the vehicle body and I don't know how to generate wheel's body

[image]