carla-simulator/carla #8812: ShaderCompileWorker failed during CarlaUnreal build process

fjmttty, 2025-04-03:
I encountered an issue while building and launching CarlaUnreal using Unreal Engine 5.5.0. The build process took a long time (~28206 seconds), and during the launch phase, ShaderCompileWorker failed with an error message stating that WorkerOutputOnly.out size is 0. The error message suggests a possible disk space issue, but I have verified that there is sufficient disk space available.

Steps to Reproduce:

Clone the Carla repository and set up the Unreal Engine environment.

Run the build command:
`cmake --build . --target launch`

Observe the Unreal Engine launch process.

An error window appears, stating that ShaderCompileWorker failed.

Expected Behavior:
The Unreal Editor should launch without ShaderCompileWorker failure.

Actual Behavior:

Unreal Editor starts launching but fails at 39% shader compilation.

The following error message is displayed:
Output file: C:/Users/***/AppData/Local/Temp/UnrealShaderWorkingDir/DF15D3345CFA7E67539A5EE690DFFF/6/WorkerOutputOnly.out size is 0.  
Are you out of disk space?

System Information:

OS: Windows 11

Unreal Engine Version: 5.5.0

Build Environment: VS 2022 Native Tools Command Prompt

GPU: [Specify your GPU model]

Disk Space: [Specify available space]

Additional Context:

I have verified that I have sufficient disk space.

The issue persists even after restarting my system.

Blyron, 2025-04-03:
This is UE5 issue, we cannot help so much there. try to delete the folders in %appdata%/local/unrealengine .
and do a git clean -fdx in carla repo

Blyron, 2025-04-03:
Also make sure carla content is there