carla-simulator/carla #8814: Compile Unreal Engine failed during Carla windows build process

Eruledge, 2025-04-04:
**Setup**
  * CARLA version: 0.9.15
  * Platform: Windows 11
  * Python version: 3.10
  * GPU: RTX 3090
  * GPU Drivers: 572.60

**Describe the bug**
When I wanted to recompile according to the Carla build documentation on my computer where Carla UE4 had been built before (but deleted), while installing Unreal Engine, I got a ".dll file corruption" error when trying the step 4 of "Unreal Engine" in the "Windows build" documentation of Carla (opening UE4Editor.exe).

**Steps to reproduce**
Follow each step of the building documentation, no obvious errors before launching the executable "Engine\Binaries\Win64\UE4Editor.exe".

**Expected behavior**
Try to uninstall Epic Games (and all related unreal engine), Steam and Steam VR (because this computer has been used by Carla to do some VR experiments.), and recompile (rebuild) unreal engine many times. But still occur same error (see screenshot below).

**Screenshots**
[image]
Everything is fine up until step 4 of "unreal engine" in the build documentation, this error occurs when trying to launch the editor .exe file in Unreal Engine after compiling.

Blyron, 2025-04-04:
This issue is related to Unreal, not to us. try to look at Epic Games forum, in a quick search I have foundt his;
https://forums.unrealengine.com/t/strange-errors-preventing-ue5-starting-dxil-dll-is-either-not-designed-to-run-on-windows-or-it-contains-an-error/739621/5

You may need to update your windows version. or rebuild

https://forums.unrealengine.com/t/projectname-4603-dll-is-either-not-designed-to-run-on-windows-or-contains-an-error/437380

Eruledge, 2025-04-04:
Thanks for your reply :) 

The first forum may not help us since the measures on it rely on Epic Games (and the "verify" function in it), but this UE cannot be detected by Epic Games.

I have tried cloning and rebuilding this UE multiple times, but it is still unavailable.

My Windows version is the newest, which does not need to be updated.

Interestingly, repeating the same build steps on my laptop works fine, but it fails and causes the bug above on my desktop.

Blyron, 2025-04-04:
As expected the issue is local to your system. Maybe permissions or something like that. Sorry but we are not gonna be help to you. Ask on the Unreal forum. We haven't seen this.

CAGTB, 2025-05-02:
或许是文件夹的权限有问题，右键文件夹，点击属性，取消只读，应用。我这边这么做了之后搞定了

Blyron, 2025-05-03:
> 或许是文件夹的权限有问题，右键文件夹，点击属性，取消只读，应用。我这边这么做了之后搞定了

In English please.