carla-simulator/carla #8829: Cant spawn Prop: Spawn failed because of invalid actor description

batuhanayfer, 2025-04-10:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10
  * Platform: windows
  * Python version: 3.8
  * GPU: RTX3070Ti

**Describe the bug**
Getting "LogCarlaServer: Responding error: Spawn failed because of invalid actor description" while trying to spawn traffic cones

**Steps to reproduce**
Try to spawn traffic cone with python API

**Logs**
LogTemp: Loaded OpenDrive file 'D:/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Maps/OpenDrive/Town10HD_Opt.xodr'
LogCarla: Spawning actor 'vehicle.mini.cooper'
LogCarla: Door name: SM_DoorL_Mini2021
LogCarla: Door name: SM_DoorR_Mini2021
LogCarla: Spawning actor 'static.prop.trafficcone01'
LogCarla: Error: Actor Description Class is null.
LogCarla: Warning: Failed to spawn actor 'static.prop.trafficcone01'
LogCarla: Error: Actor not Spawned
LogCarlaServer: Responding error: Spawn failed because of invalid actor description

Blyron, 2025-04-14:
Which steps have you followed? Are you latest in ue5-dev branch of content and code?

dedoogong, 2025-05-19:
I met the same problem in Ubuntu. 
I just packaged Carla ue5 in ubuntu and run the packaged CarlaUnreal like
```
/media/ext/Linux$ ./CarlaUnreal.sh -carla-server -log
```
it shows the carla correctly but when I run manual_control.py in another terminal,
the calra log shows 
```
[2025.05.19-02.52.27:245][842]LogCarlaServer: Responding error: Spawn failed because of invalid actor description
[2025.05.19-02.52.27:255][843]LogCarla: Spawning actor 'vehicle.fuso.mitsubishi'
[2025.05.19-02.52.27:255][843]LogCarla: Error: Actor Description Class is null.
[2025.05.19-02.52.27:255][843]LogCarla: Warning: Failed to spawn actor 'vehicle.fuso.mitsubishi'
[2025.05.19-02.52.27:255][843]LogCarla: Error: Actor not Spawned
```
repeatedly.

Blyron, 2025-08-04:
You cannot spawn BP like that. You have to spawn only static meshes

Blyron, 2025-08-04:
the name is wrong I think it would be becuase of that