carla-simulator/carla #8839: Provide make LibCarla for Carla 0.10.0

sshawn9, 2025-04-15:
**Is your feature request related to a problem? Please describe.**
I have successfully compiled Carla 0.10.0 from source. However, unlike previous versions, this version does not provide the `make LibCarla` option, which prevents me from compiling the Carla client and performing the installation.

**Describe the solution you'd like**
I would like the current version to include the `make LibCarla` compile option as it was available in earlier versions. This would allow users to compile the Carla client and install it as before.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
I deeply appreciate the work that has gone into developing Carla and sincerely hope that this enhancement can be implemented in a future release. Your efforts in supporting the community are highly valued, and I eagerly look forward to this improvement. Thank you for your consideration.

Blyron, 2025-05-06:
You can use cmake --build Build --target carla-help to look into the possible targets. The target you are looking for is carla-server and carla-client

sshawn9, 2025-05-08:
> You can use cmake --build Build --target carla-help to look into the possible targets. The target you are looking for is carla-server and carla-client

I have tried this, and it says no work to do, and I still cannot get carla-client library, wondering and looking for help.

```bash
carla@star-x:~/CarlaUE5$ cmake --build Build --target carla-help
[0/2] Re-checking globbed directories...
[1/1] cd /home/carla/CarlaUE5/Build && /opt/cmake-3.28.3-linux-x86_64/bin/cmake -E cat /home/carla/CarlaUE5/Build/Help.md
# CARLA - CMake Help
## CMake Targets
- clean
	- Type: Builtin
	- Description: Removes all build directories and files.
- carla-server
	- Type: Library
	- Description: Build the CARLA server.
- carla-client
	- Type: Library
	- Description: Build the CARLA client.
- carla-ros2-native
	- Type: CustomTarget
	- Description: Build the ROS2-Native CARLA subproject.
- carla-python-api
	- Type: CustomTarget
	- Description: Build the CARLA Python API.
- carla-python-api-install
	- Type: CustomTarget
	- Description: Build & install the CARLA Python API
- carla-unreal
	- Type: CustomTarget
	- Description: Build the CarlaUnreal subproject.
- carla-unreal-editor
	- Type: CustomTarget
	- Description: Build the CarlaUnrealEditor subproject.
- carla-unreal-package
	- Type: CustomTarget
	- Description: Create a CARLA package in default (shipping) mode.
- package
	- Type: CustomTarget
	- Description: Create a CARLA package in default (shipping) mode.
- carla-unreal-package-shipping
	- Type: CustomTarget
	- Description: Create a CARLA package in Shipping mode.
- package-shipping
	- Type: CustomTarget
	- Description: Create a CARLA package in Shipping mode.
- carla-unreal-package-debug
	- Type: CustomTarget
	- Description: Create a CARLA package in Debug mode.
- package-debug
	- Type: CustomTarget
	- Description: Create a CARLA package in Debug mode.
- carla-unreal-package-debuggame
	- Type: CustomTarget
	- Description: Create a CARLA package in DebugGame mode.
- package-debuggame
	- Type: CustomTarget
	- Description: Create a CARLA package in DebugGame mode.
- carla-unreal-package-development
	- Type: CustomTarget
	- Description: Create a CARLA package in Development mode.
- package-development
	- Type: CustomTarget
	- Description: Create a CARLA package in Development mode.
- carla-unreal-package-test
	- Type: CustomTarget
	- Description: Create a CARLA package in Test mode.
- package-test
	- Type: CustomTarget
	- Description: Create a CARLA package in Test mode.
[... 172 lines cut ...]
	- Default: "2.11.2"
- CARLA_FASTDDS_TAG
	- Description: Target Fast-DDS git tag.
	- Default: "2.11.2"
- CARLA_FASTCDR_VERSION
	- Description: Target Fast-CDR version.
	- Default: "1.1.x"
- CARLA_FASTCDR_TAG
	- Description: Target Fast-CDR git tag.
	- Default: "1.1.x"
- CARLA_FOONATHAN_MEMORY_VENDOR_VERSION
	- Description: Target foonathan_memory_vendor version.
	- Default: "master"
- CARLA_FOONATHAN_MEMORY_VENDOR_TAG
	- Description: Target foonathan_memory_vendor git tag.
	- Default: "master"
- ENABLE_DIRECTORY_CLEAN
	- Description: Enable ADDITIONAL_CLEAN_FILES on directory level
	- Default: ON

carla@star-x:~/CarlaUE5$ cmake --build Build --target carla-client
[0/2] Re-checking globbed directories...
ninja: no work to do.
carla@star-x:~/CarlaUE5$ 
```

Blyron, 2025-05-08:
That means it is already built. 
You can find the .lib in Build folder

[image]

sshawn9, 2025-05-08:
> That means it is already built. You can find the .lib in Build folder
> 
> [image]

[image]

Yes, I find it as above, and I still wondering how to get correct header files, please see the picture below, are these right header files, thanks a lot.

[image]