carla-simulator/carla #8842: carla ue5.5 runs strange

yeyueluke, 2025-04-17:
os: ubuntu22.04
GPU: 4090 24G
carla branch: ue5-dev
ue branch:ue5-dev-carla

I compile carla according to https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/
and change the Options.cmake to make 
ENABLE_PYTORCH , 
ENABLE_OSM2ODR, 
BUILD_OSM_WORLD_RENDERER, 
ENABLE_STREETMAP 
on
and compile with command :./CarlaSetup.sh --interactive --launch
 and it works ,but when i run cppclient examples ,the UI in ue works strange ,the UI looks like in picture below

[image]

Blyron, 2025-04-17:
Please, use provided Issue templates. Re-open issue with the template

closed 2025-04-17 by Blyron

yeyueluke, 2025-04-18:
ok, new issuse:https://github.com/carla-simulator/carla/issues/8846