carla-simulator/carla #8843: Where to find the carla assets to download mannually

Ceadol, 2025-04-17:
want to build carla 0.9.14 or 0.9.15, but the amazon web service has discontinued to provide the carla asset. If anyone could tell me where I can find the asset.

The following attempt has failed:

PS D:\ProgramFiles\carla-0.9.14> .\Update.bat
20221201_5ec9328
http://carla-assets.s3.amazonaws.com/20221201_5ec9328.tar.gz
Downloading "http://carla-assets.s3.amazonaws.com/20221201_5ec9328.tar.gz"...
使用“2”个参数调用“DownloadFile”时发生异常:“远程服务器返回错误: (403) 已禁止。”
所在位置 行:1 字符: 1
+ (New-Object System.Net.WebClient).DownloadFile('http://carla-assets.s ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : WebException

 Exiting with error...

open the link to the asset, then the interface displays the following:

<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>R10AG6RWGA6XCT45</RequestId>
<HostId>oWhhtxXrXkQB6v+M+k7/jClI9EVC304cb88w8nIvIBRtE52YnoApn7T0ERkDRMzjX6lK+1diJHQ=</HostId>
</Error>

GoodarzMehr, 2025-04-17:
Is [this](https://bitbucket.org/carla-simulator/carla-content/src/master/) what you are looking for?

Blyron, 2025-04-17:
change url from 
http://carla-assets.s3.amazonaws.com/20221201_5ec9328.tar.gz
to https://carla-assets.s3.us-east-005.backblazeb2.com/20221201_5ec9328.tar.gz

closed 2025-04-17 by Blyron