carla-simulator/carla #8844: Multiple Trees Merged as One Instance in CARLA

adamkanjo, 2025-04-17:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10.0
  * Platform: Ubuntu 20.04
  * Python version: 0.9.15
  * GPU: NVIDIA Corporation TU104GL [Tesla T4]
  * GPU Drivers: 535.230.02

**Describe the bug**
I have written a script to where it creates a different color for each instance vegetation segmentation object just for visual inspection, but as shown in the picture, many trees are considered as one instance ): why does this happen ? is there something wrong in my code ? any idea ?

[image]

**Steps to reproduce**
Explain the steps needed to reproduce the issue.

Run the script 
`
  import carla
  import random
  import queue
  import os
  import numpy as np
  import cv2

def main():
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()

    # Spawn vehicle
    ego_bp = blueprint_library.find('vehicle.dodge.charger')
    ego_bp.set_attribute('role_name', 'ego')
    if ego_bp.has_attribute('color'):
        ego_color = random.choice(ego_bp.get_attribute('color').recommended_values)
        ego_bp.set_attribute('color', ego_color)
    spawn_point = random.choice(world.get_map().get_spawn_points())
    vehicle = world.try_spawn_actor(ego_bp, spawn_point)
    if vehicle is None:
        print("Failed to spawn the vehicle.")
        return

    # Setup instance segmentation camera
    cam_bp = blueprint_library.find('sensor.camera.instance_segmentation')
    cam_bp.set_attribute('image_size_x', '800')
    cam_bp.set_attribute('image_size_y', '600')
    cam_bp.set_attribute('fov', '90')
    cam_transform = carla.Transform(carla.Location(x=2.0, z=1.0))
    instance_camera = world.spawn_actor(cam_bp, cam_transform, attach_to=vehicle)

    image_queue = queue.Queue()
    instance_camera.listen(image_queue.put)

    world.tick()

    try:
        instance_image = image_queue.get(timeout=10.0)

        raw_data = np.frombuffer(instance_image.raw_data, dtype=np.uint8)
        raw_data = raw_data.reshape((instance_image.height, instance_image.width, 4))
        rgb_image = raw_data[:, :, :3][:, :, ::-1]  # Convert BGRA to RGB

        height, width = rgb_image.shape[:2]
        output_image = np.zeros((height, width, 3), dtype=np.uint8)

        instance_colors = {}
        for y in range(height):
            for x in range(width):
                r, g, b = rgb_image[y, x]
                semantic_tag = r
                instance_id = (g << 8) + b

                if semantic_tag != 9:
                    continue

                if instance_id not in instance_colors:
                    # Assign a unique random color to each instance
                    instance_colors[instance_id] = tuple(np.random.randint(0, 256, size=3).tolist())

                output_image[y, x] = instance_colors[instance_id]

        # Save colored image
        output_dir = "_out"
        os.makedirs(output_dir, exist_ok=True)
        output_path = os.path.join(output_dir, "colored_trees_only.png")
        cv2.imwrite(output_path, output_image)
        print(f"Saved: {output_path}")

    except queue.Empty:
        print("No image received in time.")
    finally:
        instance_camera.stop()
        instance_camera.destroy()
        vehicle.destroy()

if __name__ == "__main__":
    main()

`

[instance.txt](https://github.com/user-attachments/files/19790735/instance.txt)

Blyron, 2025-04-17:
Quick answer for this;
Yes, multiple geometry can be the same instance as we are using HierarchicalStaticMeshComponents which allows to render multiple geometry instance in differente places for a better performance. So this is expected behaviour we could improve

Blyron, 2025-04-17:
No, there is no way right now.

Blyron, 2025-04-17:
You would need to do that in Level editor within Unreal

adamkanjo, 2025-04-17:
Thank you for the info :)

Blyron, 2025-04-17:
Closing this issue as questions has been solved.

closed 2025-04-17 by Blyron