carla-simulator/carla #8849: Launched from Editor can't switch map/level

Uzukidd, 2025-04-17:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: Windows 11
  * Python version: 0.3.12
  * GPU: RTX 4060TI
  * GPU Drivers: 561.03

**Describe the bug**
I've tried launching CARLA both from the button in the top-right corner and via the Project Launcher. However, in either case, the launched game doesn't allow me to switch maps or levels using the Python API. It's quite inconvenient to have to manually switch to the desired map or level before launching the game.
Does anyone know if I'm launching the game incorrectly, or have any suggestions? 

**Steps to reproduce**
Click launch or project launcher

**Expected behavior**
Able to switch map using Python API

**Logs**
[CarlaUE4.log](https://github.com/user-attachments/files/19792528/CarlaUE4.log)

**Scripts**
[scenario_runner](https://github.com/carla-simulator/scenario_runner)

**Screenshots**
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

Blyron, 2025-04-17:
Have you tried to press play instead of using the launcher?

Uzukidd, 2025-04-18:
> Have you tried to press play instead of using the launcher?

Thank you very much! Now it works perfectly in play mode.
Additionally, the launcher's level-switching problem is solved by setting the 'Data Build' option to 'By the book'.

[image]

closed 2025-04-18 by Uzukidd