carla-simulator/carla #8852: [Compiler] VehicleActorFactory.VehiclesParams is not blueprint visible (BlueprintReadOnly or BlueprintReadWrite). Please fix mark up or cease accessing as this will be made an error in a future release

zmd-cc, 2025-04-18:
https://github.com/carla-simulator/carla/blob/db1b4fd17673da9c41cde1c0fbfd861c99290b52/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Actor/Factory/VehicleActorFactory.cpp#L24

```
TArray<FActorDefinition> AVehicleActorFactory::GetDefinitions()
{
  LoadVehicleParametersArrayFromFile("VehicleParameters.json", VehiclesParams);
  FString UniqueVehicleParameters = GetWorld()->GetMapName().Mid(GetWorld()->StreamingLevelsPrefix.Len()) + "/Vehicles.json";
  LoadVehicleParametersArrayFromFile(UniqueVehicleParameters, MineVehiclesParams);
  for (const FVehicleParameters& VehicleParams : MineVehiclesParams)
  {
    VehiclesParams.Add(VehicleParams);
  }
  UActorBlueprintFunctionLibrary::MakeVehicleDefinitions(VehiclesParams, Definitions);
  return Definitions;
}
```

this line use VehicleParams in the for-loop which has the same name as the class protected member 'VehicleParams'

when I ran `cmake --build Build/Linux-Release/ --target package`
errors show up

```
LogInit: Display: LogBlueprint: Error: [AssetLog] /home/Carla/carla-ue5-dev/Unreal/CarlaUnreal/Content/Carla/Blueprints/Vehicles/BP_VehicleFactory.uasset: [Compiler] VehicleActorFactory.VehiclesParams is not blueprint visible (BlueprintReadOnly or BlueprintReadWrite). Please fix mark up or cease accessing as this will be made an error in a future release.  Get VehiclesParams from Source: /Game/Carla/Blueprints/Vehicles/BP_VehicleFactory.BP_VehicleFactory
LogInit: Display: LogBlueprint: Error: [AssetLog] /home/Carla/carla-ue5-dev/Unreal/CarlaUnreal/Content/Carla/Blueprints/Vehicles/BP_VehicleFactory.uasset: [Compiler] VehicleActorFactory.MineVehiclesParams is not blueprint visible (BlueprintReadOnly or BlueprintReadWrite). Please fix mark up or cease accessing as this will be made an error in a future release.  Get MineVehiclesParams from Source: /Game/Carla/Blueprints/Vehicles/BP_VehicleFactory.BP_VehicleFactory
LogInit: Display: LogBlueprint: Error: [AssetLog] /home/Carla/carla-ue5-dev/Unreal/CarlaUnreal/Content/Carla/Blueprints/Vehicles/BP_VehicleFactory.uasset: [Compiler] VehicleActorFactory.VehiclesParams is not blueprint writable.  Set VehiclesParams from Source: /Game/Carla/Blueprints/Vehicles/BP_VehicleFactory.BP_VehicleFactory
LogInit: Display: LogBlueprint: Error: [AssetLog] /home/Carla/carla-ue5-dev/Unreal/CarlaUnreal/Content/Carla/Blueprints/Vehicles/BP_VehicleFactory.uasset: [Compiler] VehicleActorFactory.MineVehiclesParams is not blueprint writable.  Set MineVehiclesParams from Source: /Game/Carla/Blueprints/Vehicles/BP_VehicleFactory.BP_VehicleFactory
```
**Why not replace the for-loop with** ` VehiclesParams.Append(MineVehiclesParams);` ?

@glopezdiest Please check this issue

lusCombeNCut, 2025-04-30:
Hi, have you come across a fix for this? Also getting the same error.

zmd-cc, 2025-05-06:
Yes, as I said above
Go to file `carla/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Actor/Factory/VehicleActorFactory.cpp`
Replace L24-27 the for-loop 
```
for (const FVehicleParameters& VehicleParams : MineVehiclesParams)
  {
    VehiclesParams.Add(VehicleParams);
  }
```
with `VehiclesParams.Append(MineVehiclesParams);`

closed 2025-05-06 by MarcelPiNacy-CVC