carla-simulator/carla #8863: Broken PythonAPI Example

manne11, 2025-04-24:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version:  Ue5-dev branch, latest git hash a6a1189f440a
  * Platform: Ubuntu 22.04.5 
  * Python version: 3.10.12
  * GPU:  NVIDIA GeForce RTX 4060 Ti, 16Gb VRAM
  * GPU Drivers: 550.120 (nvidia)

**Describe the bug**
After successfully building carla by following instructions in https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/
. I started running the PythonAPI examples. I started the server which popped up the window with Town 10 map in it. Then when running the generate_traffic.py, I get an error:
```
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
ERROR: Spawn failed because of invalid actor description
spawned 0 vehicles and 0 walkers, press Ctrl+C to exit.

```

**Steps to reproduce**
1. cmake --build Build --target package
2. ./Build/Package/StagedBuilds/Carla-0.10.0-Linux-Shipping/Linux/CarlaUnreal.sh
3. In another terminal after installing the requirements.txt , 
    python3 Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/generate_traffic.py

**Expected behavior**
Expected Walkers and vehicles to spawn up in the pop up window after running the generate_traffic.py script

**Logs**
Unsure where the client logs are when running the python script, the server was active and did not crash.
 server was listening on the following ports:
```

 netstat -tunlp

(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:2002            0.0.0.0:*               LISTEN      4191/CarlaUnreal-Li 
tcp        0      0 0.0.0.0:2000            0.0.0.0:*               LISTEN      4191/CarlaUnreal-Li 
tcp        0      0 0.0.0.0:2001            0.0.0.0:*               LISTEN      4191/CarlaUnreal-Li 
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      -                   
```        

**Additional context**
Just created a new build and trying to check if the Python API is functional

Blyron, 2025-04-24:
We are aware of the issue.

Blyron, 2025-04-24:
We will be fixing it soon

manne11, 2025-04-24:
Thanks, will keep an eye on it, will close the issue once verified.

J160KU, 2025-05-16:
Running generate_traffic.py in CARLA 0.10.0 on Ubuntu 22.04 is spawning vehicles, but no pedestrians.

Do you know how to get pedestrians in the simulation?

J160KU, 2025-05-22:
Hi @dedoogong,

I just run the Python script, so nothing much. Latest CARLA on Windows spawns vehicles AND pedestrians, though.

Blyron, 2025-06-13:
Yeah. in latest ue5-dev branch of content and code. Everything shouyld be working now

closed 2025-08-04 by Blyron