carla-simulator/carla #8865: Not being able to build & launch the editor on Ubuntu 24.04

CarlaEnjoyer, 2025-04-24:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10.0
  * Platform: Ubuntu 24.04
  * Python version: 3.12.9
  * GPU: RTX 3090
  * GPU Drivers: 570.86.15

**Describe the bug**
When trying to build and launch the editor with the `cmake --build Build --target launch` command it throws an error when trying to `Link (lld) CarlaUnreal (UBA disabled)`
**Steps to reproduce**
- Build UE5 via the extended instructions from the documentation.
- Set the env variable and then build carla ny following the normal setup instead of the extended instructions since there is no cmake for a linux release
- configure and build carla and the python api
- try to build and launch the editor

**Expected behavior**
The Carla editor Builds and launches.

**Logs**
Log started at 04/24/2025 14:53:38 (2025-04-24T12:53:38Z)
No config file at /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Restricted/NotForLicensees/Programs/UnrealBuildTool/BuildConfiguration.xml
No config file at /usr/share/Unreal Engine/UnrealBuildTool/BuildConfiguration.xml
No config file at /home/user/snap/code/190/.local/share/Unreal Engine/UnrealBuildTool/BuildConfiguration.xml
No config file at /home/user/Documents/Unreal Engine/UnrealBuildTool/BuildConfiguration.xml
Configuration will be read from:
  /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Saved/UnrealBuildTool/BuildConfiguration.xml
  /home/user/.config/Unreal Engine/UnrealBuildTool/BuildConfiguration.xml
Setting temp directory to '/tmp/UnrealBuildTool/d8b99b33'
    Registering build platform: UnrealBuildTool.AndroidPlatformFactory
Unable to find valid SDK(s) for Android:
  Found Sdk Version, MinRequired=r25b, MaxRequired=r27.
  Found AutoSdk Version, Required=r25b.
        Registering build platform: Android - buildable: False
    Registering build platform: UnrealBuildTool.IOSPlatformFactory
        Registering build platform: IOS - buildable: False
    Registering build platform: UnrealBuildTool.LinuxPlatformFactory
Linux Installed SDK(s): Version_Sdk=v23_clang-18.1.0-rockylinux8, CurrentVersion_Sdk=v23_clang-18.1.0-rockylinux8, Version_AutoSdk=v23_clang-18.1.0-rockylinux8, CurrentVersion_AutoSdk=
        Registering build platform: Linux - buildable: True
LinuxArm64 Installed SDK(s): Version_Sdk=v23_clang-18.1.0-rockylinux8, CurrentVersion_Sdk=v23_clang-18.1.0-rockylinux8, Version_AutoSdk=v23_clang-18.1.0-rockylinux8, CurrentVersion_AutoSdk=
        Registering build platform: LinuxArm64 - buildable: True
    Registering build platform: UnrealBuildTool.MacPlatformFactory
        Registering build platform: Mac - buildable: False
    Registering build platform: UnrealBuildTool.TVOSPlatformFactory
        Registering build platform: TVOS - buildable: False
    Registering build platform: UnrealBuildTool.WindowsPlatformFactory
        Registering build platform: Win64 - buildable: False
    Registering build platform: UnrealBuildTool.VisionOSPlatformFactory
        Registering build platform: VisionOS - buildable: False
Command line: /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool/UnrealBuildTool.dll CarlaUnreal Linux Development -project=/home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw
Log file: /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Programs/UnrealBuildTool/Log.txt
Deleting old log file: /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Programs/UnrealBuildTool/Log-backup-2025.04.24-11.31.43.txt

Using 'git status' to determine working set for adaptive non-unity build (/home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine).
Waiting for 'git status' command to complete
Generated code is up to date.
CarlaUnreal: Produced item "/home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/Unreal/CarlaUnreal/Binaries/Linux/CarlaUnreal" doesn't exist.
Module.Carla.3.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:04
Module.Carla.4.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:05
Module.Carla.5.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:05
Module.Carla.6.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:04
Module.Carla.7.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:03
Module.Carla.8.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:11
Module.Carla.9.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:19
CarlaUnreal.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:05
PCH.CarlaUnreal.h: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:04:58
CarlaUnreal.target: Produced item "/home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/Unreal/CarlaUnreal/Binaries/Linux/CarlaUnreal.target" doesn't exist.
Module.Carla.1.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:04:59
PCH.Carla.h: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:04:51
Module.Carla.10.cpp: Dependency /home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/LibCarla/source/util/ue-header-guard-begin.h is newer than the last execution of the action: 04/24/2025 14:48:29 vs 04/24/2025 14:05:24
[... 43 lines cut ...]
CompilationResultException: OtherCompilationError
   at UnrealBuildTool.ActionGraph.ExecuteActionsAsync(BuildConfiguration BuildConfiguration, List`1 ActionsToExecute, List`1 TargetDescriptors, ILogger Logger, IActionArtifactCache actionArtifactCache) in /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/System/ActionGraph.cs:line 471
   at UnrealBuildTool.BuildMode.BuildAsync(TargetMakefile[] Makefiles, List`1 TargetDescriptors, BuildConfiguration BuildConfiguration, BuildOptions Options, FileReference WriteOutdatedActionsFile, ILogger Logger, String ActionTypeFilter) in /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 823
   at UnrealBuildTool.BuildMode.BuildAsync(List`1 TargetDescriptors, BuildConfiguration BuildConfiguration, ISourceFileWorkingSet WorkingSet, BuildOptions Options, FileReference WriteOutdatedActionsFile, ILogger Logger, Boolean bSkipPreBuildTargets, String ActionTypeFilter) in /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 415
   at UnrealBuildTool.BuildMode.ExecuteAsync(CommandLineArguments Arguments, ILogger Logger) in /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 255
   at UnrealBuildTool.UnrealBuildTool.Main(String[] ArgumentsArray) in /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/UnrealBuildTool.cs:line 687
WriteFileIfChanged() wrote 0 changed files of 0 requested writes.
Timeline:

[ 0.000]
[ 0.000](+145.248) <unknown>
[... cut ...]
 Link (lld) CarlaUnreal (UBA disabled): WorkingDirectory /home/user/repos/unreal_engine_carla_0.10.0/UnrealEngine/Engine/Source
 Link (lld) CarlaUnreal (UBA disabled): /bin/sh  "/home/user/repos/unreal_engine_carla_0.10.0/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Link-CarlaUnreal.link.sh"
ld.lld: 

**Additional context**
I had to install UE5 by using the extended instructions and then proceed with the normal setup but I reference the UE5 with my env variable. Also using a custom python environment.

XinHu98, 2025-05-03:
I face the same problem

qhaas, 2025-05-03:
Problem still exists on Ubuntu 24.04, [carla unreal engine fork commit eea6a8e677e3509a3e56122d69e3d380bb624bbf](https://github.com/CarlaUnreal/UnrealEngine/commit/eea6a8e677e3509a3e56122d69e3d380bb624bbf) and carla commit 83a7d44d4693bbfc2509186e1867214b64b14c12
Error:
```
[1/2] Link (lld) CarlaUnreal (UBA disabled)
ld.lld: error: undefined symbol: __isoc23_strtol
>>> referenced by server.cc
>>>               server.cc.o:(clmdep_asio::detail::socket_ops::inet_pton(int, char const*, void*, unsigned long*, std::__1::error_code&)) in archive /mnt/carla-src/Build/_deps/rpclib-build/librpc.a
>>> referenced by pugixml.cpp
>>>               pugixml.cpp.o:(pugi::impl::(anonymous namespace)::convert_number_to_string(double, pugi::impl::(anonymous namespace)::xpath_allocator*)) in archive /mnt/carla-src/Build/LibCarla/libcarla-server.a
>>> referenced by router.cpp
>>>               router.cpp.o:(boost::asio::detail::socket_ops::inet_pton(int, char const*, void*, unsigned long*, boost::system::error_code&)) in archive /mnt/carla-src/Build/LibCarla/libcarla-server.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

Dockerfile:
```
FROM Ubuntu:24.04

# use UTC as default timezone
ENV TZ=Etc/UTC
# debian-like images default to ascii
ENV LANG=C.UTF-8

# carla build apt deps
# from https://github.com/carla-simulator/carla/blob/ue5-dev/Util/SetupUtils/InstallPrerequisites.sh
ARG APT_PACKAGES="\
    build-essential \
    make \
    ninja-build \
    libvulkan1 \
    libpng-dev \
    libtiff5-dev \
    libjpeg-dev \
    tzdata \
    sed \
    curl \
    libtool \
    rsync \
    libxml2-dev \
    git \
    git-lfs \
    python3 \
    python3-dev \
    python3-pip \
    python3-build \
    python3-venv \
    cmake \
"
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y $APT_PACKAGES && apt-get clean
```

Commands (assuming UE5 is built and located at `CARLA_UNREAL_ENGINE_PATH`):
```
$ cmake -G Ninja -S . -B Build \
            --toolchain=$PWD/CMake/Toolchain.cmake \
            -DCMAKE_BUILD_TYPE=Release \
            -DENABLE_ROS2=ON \
            -DBUILD_CARLA_UNREAL=ON \
            -DCARLA_UNREAL_ENGINE_PATH=$CARLA_UNREAL_ENGINE_PATH \
            -DCARLA_UNREAL_BUILD_TYPE=Shipping
...
$ cmake --build Build
$ cmake --build Build --target package
```