carla-simulator/carla #8866: Carla 0.10.0 Instance Segmentation Camera ISM/HISM support

AleksiNarkilahti, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
CARLA 0.10.0 (UE5) Instance Segmentation Sensor currently does not support Instanced Static Mesh (ISM) or Hierarchical Instanced Static Mesh (HISM) components.

The Instance Segmentation Sensor in CARLA assigns the same ActorID and Label to all instances of ISM/HISM in the map. This results in incorrect instance identification in the output image, as all instances are grouped under the same ActorID and Label, which is not the desired behavior for instance segmentation.

**Describe the solution you'd like**
Add support for ISM/HISM to the Instance Segmentation Sensor. This will likely require modifications to the Unreal Engine source code to handle these components correctly and ensure proper instance identification.

**Describe alternatives you've considered**
As an alternative, the documentation should clearly state that ISM/HISM are not supported by the sensor. Additionally, it would be helpful to mention this limitation in the map creation process and/or sensor documentation to avoid confusion.

Bvallon-sl, 2025-05-23:
Hi,
I'm planning to implement this feature. Does someone have an idea on how this can be done?

Thanks.

Blyron, 2025-06-13:
This requires Engine modifications. Look for SegmentationPass and then on mesh processing when rendering those. You will need to modify the colors based on the instance number.

zmd-cc, 2025-07-02:
> This requires Engine modifications. Look for SegmentationPass and then on mesh processing when rendering those. You will need to modify the colors based on the instance number.

@Blyron Could you please explain SegmentationPass more specifically? I read the Tagged.cpp but still no idea how taggers work with Instance Segmentation Camera. 

Thanks