carla-simulator/carla #8867: draw_string() is not clear in PythonAPI  world.debug..clear_debug_shape()

donghaiwang, 2025-04-25:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15 ue4-dev lastest
  * Platform: windows 10
  * Python version: 3.7
  * GPU: RTX3090

**Describe the bug**
draw_box() will clear. However, draw_string() is not clear in PythonAPI  world.debug..clear_debug_shape()

**Scripts**
```
import time

import carla
import random

# 连接到客户端并获取世界对象
client = carla.Client('localhost', 2000)
world = client.get_world()

# 我们还将设置观察者，这样我们就可以看到我们在做什么
spectator = world.get_spectator()

spawn_points = world.get_map().get_spawn_points()

# 将观察者移动到俯视图
transform = carla.Transform(spawn_points[0].transform(carla.Location(x=-4, z=100)), carla.Rotation(yaw=-180, pitch=-90))
spectator.set_transform(transform)

debug_helper = world.debug

# 在地图上用数字标出生成点的位置
for i, spawn_point in enumerate(spawn_points):
    debug_helper.draw_string(spawn_point.location, str(i), life_time=15)  #

debug = world.debug
world_snapshot = world.get_snapshot()

for actor_snapshot in world_snapshot:
    actual_actor = world.get_actor(actor_snapshot.id)
    if actual_actor.type_id == 'traffic.traffic_light':
        debug.draw_box(carla.BoundingBox(actor_snapshot.get_transform().location,carla.Vector3D(0.5,0.5,2)),actor_snapshot.get_transform().rotation, 0.05, carla.Color(255,0,0,0),0)
# ...

time.sleep(3)
# 清除调式时的绘制信息
debug_helper.clear_debug_shape()
```

**Screenshots**

[image]

Blyron, 2025-04-25:
What is the expected the result?

Blyron, 2025-05-06:
PR ready to get this

donghaiwang, 2025-06-05:
After [this PR](https://github.com/carla-simulator/carla/pull/8885) modification, it seems that the function `debug_helper.clear_debug_shape()` does not take effect.

PR #8885 (2025-05-15): Load Maps and Remove Draw Strings

#### Description

This PR aims to enable load maps from different places to be loaded and not only in Carla/Maps folder of content. 
Also enables the capability of disable draw debug strings

Fixes #8867   

#### Where has this been tested?

  * **Platform(s):** Windows
  * **Python version(s):** 3.8.10
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

- - -
This change is [[image]](https://reviewable.io/reviews/carla-simulator/c
closed 2025-06-13 by Blyron