carla-simulator/carla #8893: FastDDS 2.11.2 checkout tag fails during build

sledger1, 2025-05-08:
* CARLA version: 0.10.0
  * Platform: Ubuntu 22.04
  * Python version: 3.10
  * GPU: A6000
  * GPU Drivers: 

Hello,

When attempting to buildl from the CarlaSetup.sh script I get the following error:

FAILED: Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-download /home/***/Carly/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-download 
cd /home/***/Carly/CarlaUE5/Build/Ros2Native/fastdds-prefix/src && /usr/bin/cmake -P /home/***/Carly/CarlaUE5/Build/Ros2Native/fastdds-prefix/tmp/fastdds-gitclone.cmake && /usr/bin/cmake -E touch /home/***/Carly/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-download
Cloning into 'fastdds'...
fatal: invalid reference: 2.11.2
CMake Error at /home/***/Carly/CarlaUE5/Build/Ros2Native/fastdds-prefix/tmp/fastdds-gitclone.cmake:49 (message):
  Failed to checkout tag: '2.11.2'

I see Fast-DDS repo was recently updated. Any guidance is appreciated!

Thanks,
Scott

sledger1, 2025-05-08:
Quick update - I updated the Options.cmake file to target version 2.11.3 for Fast-DDS and it seems to be working now. 

# ==== FASTDDS ====

carla_string_option (
  CARLA_FASTDDS_VERSION
  "Target Fast-DDS version."
  2.11.3
)

qhaas, 2025-06-08:
Still an issue on commit e158e267d77aaf53ce7a14d534fd8ee0ac8dea41

closed 2025-08-04 by Blyron