carla-simulator/carla #8897: Carla 0.10.0 ROS 2 IMU Coordinate System

joeyjyyang, 2025-05-13:
Hi,

I'm confused about what coordinate system the IMU sensor is publishing ROS 2 data in. I am spawning the IMU in my `.json` config like so:
```
"type": "sensor.other.imu",
"id": "imu",
"spawn_point": {"x": 0.0, "y": 0.0, "z": 0.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
"attributes": {
    "sensor_tick": 0.01
}
```

When I inspect the `linear_acceleration.y` and `angular_velocity.z` components of the IMU measurements, I can see that the values are opposite of my vehicle's direction and need to be negated (e.g. vehicle turns left but `linear_acceleration.y` and `angular_velocity.z` are negative, and vice versa). So far, this consistent with Carla using LHS while ROS uses RHS. My confusion comes from the seeing the `linear_acceleration.z` component of the IMU measurements showing `+9.81 m/s^2`, as opposed to the expected `-9.81 m/s^2`.

I'm wondering:
1. Is `enable_for_ros()` supposed to implicitly convert Carla LHS to ROS RHS for all sensors? I don't think this happening, but would like to confirm.
2. Is the IMU in fact in LHS, but Carla convention is to add the gravity constant to `linear_acceleration.z`, as opposed to subtracting it (like in ROS convention); or, does the IMU spawn in some of inverted RHS orientation and needs to be flipped?

joeyjyyang, 2025-05-13:
Okay it looks like I'm wrong. As per [REP 145 in ROS](https://www.ros.org/reps/rep-0145.html), the ROS convention is to add the gravity constant to `linear_acceleration.z`, resulting in `+9.81 m.s^2`. This in conjunction with the observations in my previous message indicate that the Carla IMU is publishing measurements to ROS in LHS. I'm going to create a quick node in my autonomy stack to convert the IMU measurements into RHS.

Blyron, 2025-05-15:
Please use the provided templates when opening issues. Closing it. Feel free to reopen with a proper template

closed 2025-05-15 by Blyron