carla-simulator/carla #8906: Bug Simple Scenic3 script from Fork with Carla 0.10

davidscientistcom, 2025-05-14:
## 🛠️ Setup

Describe the setup you are using to run CARLA along with its version:

- **CARLA version:** 0.10.0 (running on Unreal Engine 5)
- **Scenic version:** Latest commit from the fork [`carla-simulator/Scenic`](https://github.com/carla-simulator/Scenic)
- **Platform:** Windows 11 Pro (fully updated, build 23H2)
- **Python version:** 3.12.9 (installed and isolated via Conda environment)
- **GPU:** NVIDIA RTX 3060
- **GPU Drivers:** Latest available through GeForce Experience (version 551.xx)

---

## 🐛 Describe the bug

When trying to simulate a basic Scenic scenario using the CLI with `--2d --simulate`, the process fails during the **pruning phase**, specifically when Scenic attempts to compute the `planarInradius` of a `Car` object.

The issue occurs inside `scenic/core/regions.py` when the method `makeShapelyPoint(...)` is called with a **tuple instead of two positional arguments**.

This leads to the following error:

This appears to be due to the incorrect call:

```python
makeShapelyPoint(((minx + maxx) / 2, (maxy + miny) / 2))

```

Instead something like:
makeShapelyPoint(*((minx + maxx)/2, (maxy + miny)/2))

## Steps to reproduce
git clone https://github.com/carla-simulator/Scenic.git
cd Scenic

conda create -n carla10 python=3.12
conda activate carla10

pip install -e .

pip install carla-0.10.0-cp312-cp312-win_amd64.whl

scenic examples/carla/car.scenic --2d --simulate -b

## Pip Libraries:
 pip list
Package                Version     Editable project location
---------------------- ----------- --------------------------------
antlr4-python3-runtime 4.13.2
attrs                  25.3.0
carla                  0.10.0
contourpy              1.3.2
cycler                 0.12.1
Cython                 3.1.0
dotmap                 1.3.30
fonttools              4.58.0
imageio                2.37.0
kiwisolver             1.4.8
lazy_loader            0.4
manifold3d             2.3.0
mapbox_earcut          1.0.3
matplotlib             3.10.3
networkx               3.4.2
numpy                  1.26.4
opencv-python          4.11.0.86
packaging              25.0
pegen                  0.3.0
pillow                 11.2.1
pip                    25.1
pygame                 2.6.1
pyglet                 1.5.31
pyparsing              3.2.3
python-dateutil        2.9.0.post0
python-fcl             0.7.0.8
rtree                  1.4.0
rv-ltl                 0.1.0
scenic                 3.0.0       C:\Users\python\Documents\Scenic
scikit-image           0.25.2
scipy                  1.15.3
setuptools             78.1.1
shapely                2.1.0
six                    1.17.0
tifffile               2025.5.10
trimesh                4.6.9
wheel                  0.45.1
## Log Output
Beginning scenario construction...
pygame 2.6.1 (SDL 2.28.4, Python 3.12.9)
Hello from the pygame community. https://www.pygame.org/contribute.html
  Loading cached version of road network...
  Loaded cached network in 0.12 seconds.
  Pruning scenario...
Traceback (most recent call last):
  ...
  File "scenic/core/regions.py", line XXXX, in inradius
    center = makeShapelyPoint(((minx + maxx) / 2, (maxy + miny) / 2))
TypeError: points() takes from 2 to 3 positional arguments but 1 were given

J160KU, 2025-05-16:
Hello @davidscientistcom,

do you know if Scenic 3 can do everything Scenario Runner also can regarding creating traffic scenarios? I need to consider to switch from Scenario Runner, because it is not available (yet?) for CARLA 0.10.0

M-Colley, 2026-10-24:
This comes from the Scenic fork's shared point helper. At the fork's current pinned revision [`9b9497b7`, geometry.py:113](https://github.com/carla-simulator/Scenic/blob/9b9497b7a90eb3e8c41dafa7f8248f632c3a4cde/src/scenic/core/geometry.py#L113), `makeShapelyPoint` still aliases the private `shapely.lib.points` function.

Upstream Scenic already fixed this in [PR336](https://github.com/Scenic-Foundation/Scenic/pull/336), merged as [`b72430ca`](https://github.com/Scenic-Foundation/Scenic/commit/b72430ca367ea37dc42932e77fe3aa53998ab5bd): change that alias to the public [`shapely.points`](https://shapely.readthedocs.io/en/stable/reference/shapely.points.html) API. This fixes the shared helper rather than changing individual callers to unpack their arguments.

I reproduced the reported helper error with installed Shapely2.1.2; the upstream replacement accepts the same `(0.5, 0.5)` input and returns a Point. The CARLA Scenic fork still contains the old alias, so the fork remains affected. This check covered the helper only; I have not validated the complete Scenic scenario or a CARLA server run.