carla-simulator/carla #8914: Can't connect to the packaged Carla-eu5 from python client

dedoogong, 2025-05-16:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10.0 unreal 5.5 built from source 
  * Platform: Windows 11
  * Python version: 3.10
  * GPU: 4090
  * GPU Drivers: 532

**Describe the bug**
After setup my custom carla level based on the provided town map, I packaged the project to get the Windows/CarlaUnreal.exe. I launched the exe file like " D:\Windows> .\CarlaUnreal.exe -carla-server -carla-rpc-port=2000 -world-port=2000 -log" and I tried to run python example like manual_control.py but it shows "world = client.get_world() RuntimeError"

**Steps to reproduce**
described above

**Expected behavior**
described above

**Logs**
described above
  * Unreal Engine 4: _Unreal/CarlaUE4/Saved/Logs_
  * Unreal Engine 5: _Unreal/CarlaUnreal/Saved/Logs_

**Scripts**
described above

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image] 
[image]

**Additional context**
Add any other context about the problem here.

dedoogong, 2025-05-18:
for reference, here is log but I cant find any port related info from it

```
Log file open, 05/18/25 15:05:13
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogProfilingDebugging: Loading WinPixEventRuntime.dll for PIX profiling (from ../../../Engine/Binaries/ThirdParty/Windows/WinPixEventRuntime/x64).
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogWindows: Enabling Tpause support
LogWindows: Warning: Failed to set completion port for job object "UE.ShaderCompileWorker.JobGroup": �Ű� ������ Ʋ���ϴ�.
LogWindows: Custom abort handler registered for crash reporting.
LogCore: Display: UTS: The Unreal Trace Server binary is not available ('../../../Engine/Binaries/Win64/UnrealTraceServer.exe')
LogTrace: Initializing trace...
LogTrace: Finished trace initialization.
LogCsvProfiler: Display: Metadata set : platform="Windows"
LogCsvProfiler: Display: Metadata set : config="Debug"
LogCsvProfiler: Display: Metadata set : buildversion="UE5-CL-0"
LogCsvProfiler: Display: Metadata set : engineversion="5.5.0-0+UE5"
LogCsvProfiler: Display: Metadata set : os="Windows 11 (24H2) [10.0.26100.3476] "
LogCsvProfiler: Display: Metadata set : cpu="GenuineIntel|12th Gen Intel(R) Core(TM) i9-12900K"
LogCsvProfiler: Display: Metadata set : pgoenabled="0"
LogCsvProfiler: Display: Metadata set : pgoprofilingenabled="0"
LogCsvProfiler: Display: Metadata set : ltoenabled="0"
LogCsvProfiler: Display: Metadata set : asan="0"
LogCsvProfiler: Display: Metadata set : commandline="" CarlaUnreal -carla-server -log""
LogCsvProfiler: Display: Metadata set : loginid="dca7c44c4395ec760b04f981a1ac9cf8"
LogCsvProfiler: Display: Metadata set : llm="0"
LogPakFile: Initializing PakPlatformFile
LogIoDispatcher: Display: Reading toc: ../../../CarlaUnreal/Content/Paks/global.utoc
LogIoDispatcher: Display: Toc loaded : ../../../CarlaUnreal/Content/Paks/global.utoc, Id=ffffffffffffffff, Order=0, EntryCount=1, SignatureHash=0000000000000000000000000000000000000000
LogIoDispatcher: Display: Mounting container '../../../CarlaUnreal/Content/Paks/global.utoc' in location slot 0
LogPakFile: Display: Initialized I/O dispatcher file backend. Mounted the global container: ../../../CarlaUnreal/Content/Paks/global.utoc
LogPakFile: Display: Found Pak file ../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.pak attempting to mount.
LogPakFile: Display: Mounting pak file ../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.pak.
LogIoDispatcher: Display: Reading toc: ../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.utoc
LogIoDispatcher: Display: Toc loaded : ../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.utoc, Id=83900d0c0c6e7ddd, Order=3, EntryCount=17, SignatureHash=0000000000000000000000000000000000000000
LogIoDispatcher: Display: Mounting container '../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.utoc' in location slot 0
LogPakFile: Display: Mounted IoStore container "../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.utoc"
LogFilePackageStore: Mounting container: Id=83900d0c0c6e7ddd, Order=3, NumPackages=0
LogPakFile: Display: Mounted Pak file '../../../CarlaUnreal/Content/Paks/pakchunk0optional-Windows.pak', mount point: '/'
LogPakFile: Display: Found Pak file ../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.pak attempting to mount.
LogPakFile: Display: Mounting pak file ../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.pak.
LogIoDispatcher: Display: Reading toc: ../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.utoc
LogIoDispatcher: Display: Toc loaded : ../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.utoc, Id=81e4dd3d9595e447, Order=3, EntryCount=10175, SignatureHash=0000000000000000000000000000000000000000
LogIoDispatcher: Display: Mounting container '../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.utoc' in location slot 0
LogPakFile: Display: Mounted IoStore container "../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.utoc"
LogFilePackageStore: Mounting container: Id=81e4dd3d9595e447, Order=3, NumPackages=6255
LogPakFile: Display: Mounted Pak file '../../../CarlaUnreal/Content/Paks/pakchunk0-Windows.pak', mount point: '../../../'
LogStats: Stats thread started at 0.427836
LogIris: FNetObjectFactoryRegistry::UnregisterFactory is unregistering factory: None name: NetActorFactory id: 0
LogIris: FNetObjectFactoryRegistry::UnregisterFactory is unregistering factory: None name: NetSubObjectFactory id: 1
LogICUInternationalization: OS requested locale 'ko-KR' is not supported. Using the OS requested language of 'ko-KR' as the locale.
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +9:00, Platform Override: ''
LogInit: Session CrashGUID >====================================================
         Session CrashGUID >   UECC-Windows-A63B219A4521BBB8BE0F1DA0DCEB96AA
         Session CrashGUID >====================================================
LogAssetRegistry: Premade AssetRegistry loaded from '../../../CarlaUnreal/AssetRegistry.bin'
LogConfig: No local boot hotfix file found at: [../../../CarlaUnreal/Saved/PersistentDownloadDir/HotfixForNextBoot.txt]
LogAudio: Display: Pre-Initializing Audio Device Manager...
LogAudio: Display: AudioInfo: 'OPUS' Registered
LogAudioDebug: Display: Lib vorbis DLL was dynamically loaded.
[... cut ...]
[2025.05.18-06.07.00:478][748]LogStreaming: Warning: Failed to read file 'F:/3/Windows/CarlaUnreal/Content/../../../CarlaUnreal/Content/Carla/Maps/Nav/Town10HD_Opt.bin' error.
[2025.05.18-06.08.04:788][455]LogCarla: Sensor 1 '' created in primary server
[2025.05.18-06.08.04:941][455]LogStreaming: Warning: Failed to read file 'F:/3/Windows/CarlaUnreal/Content/../../../CarlaUnreal/Content/Carla/Maps/Nav/Town10HD_Opt.bin' error.
[2025.05.18-06.08.11:035][455]LogCarla: New secondary connection detected
[2025.05.18-06.08.23:951][455]LogCore: Engine exit requested (reason: ConsoleCtrl RequestExit)

```

eldar, 2025-06-09:
I'm also getting the errors of this type:

```
LogCarla: Error: Actor_0 has no SM assigned to the ISM
```

How could I fix it?