carla-simulator/carla #8923: Can't run manual_control.py with Carla-ue5 in ubuntu

dedoogong, 2025-05-19:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.10.0 ue5
  * Platform: Ubuntu 22.04
  * Python version: 3.8
  * GPU: 4090
  * GPU Drivers: 560.35.05

**Describe the bug**
After pressing play button in Carla-UE5 editor, I run manual_control.py in a terminal. 
After some seconds,  Carla-UE5 just crashed. 

**Steps to reproduce**
same as above
 
**Logs**

[2025.05.19-03.06.08:199][582]LogStaticMesh: Clustering [8.79s]. Ratio: 1.017593
[2025.05.19-03.06.08:264][582]LogStaticMesh: Leaves [0.07s]
[2025.05.19-03.06.08:313][582]LogStaticMesh: Leaves [0.13s]
Assertion failed: ExternalEdgeOffset == ExternalEdges.Num() [File:./Developer/NaniteBuilder/Private/ClusterDAG.cpp] [Line: 126] 

Signal 11 caught.
Malloc Size=262146 LargeMemoryPoolOffset=262162 
CommonUnixCrashHandler: Signal=11
[2025.05.19-03.06.09:384][582]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x0000000000000003

[2025.05.19-03.06.09:384][582]LogCore: Assertion failed: ExternalEdgeOffset == ExternalEdges.Num() [File:./Developer/NaniteBuilder/Private/ClusterDAG.cpp] [Line: 126] 

0x00007caefe356ea2 libUnrealEditor-NaniteBuilder.so!Nanite::BuildDAG(TArray<Nanite::FClusterGroup, TSizedDefaultAllocator<32> >&, TArray<Nanite::FCluster, TSizedDefaultAllocator<32> >&, unsigned int, unsigned int, unsigned int, TBounds<float>&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/NaniteBuilder/Private/ClusterDAG.cpp:126]
0x00007caefe361990 libUnrealEditor-NaniteBuilder.so!Nanite::FBuilderModule::Build(Nanite::FResources&, Nanite::IBuilderModule::FInputMeshData&, Nanite::IBuilderModule::FOutputMeshData*, FMeshNaniteSettings const&, TDelegate<void (bool), FDefaultDelegateUserPolicy>) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/NaniteBuilder/Private/NaniteBuilder.cpp:634]
0x00007cafbe0e729b libUnrealEditor-MeshBuilder.so!FSkeletalMeshBuilder::Build(FSkeletalMeshBuildParameters const&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/MeshBuilder/Private/SkeletalMeshBuilder.cpp:181]
0x00007cafbe0e5d6a libUnrealEditor-MeshBuilder.so!FMeshBuilderModule::BuildSkeletalMesh(FSkeletalMeshBuildParameters const&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/MeshBuilder/Private/MeshBuilderModule.cpp:81]
0x00007cafd4a6768b libUnrealEditor-Engine.so!USkeletalMesh::BuildLODModel(ITargetPlatform const*, int) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:6095]
0x00007cafd4ae3a36 libUnrealEditor-Engine.so!FSkeletalMeshRenderData::Cache(ITargetPlatform const*, USkinnedAsset*, FSkinnedAssetCompilationContext*) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMeshRenderData.cpp:316]
0x00007cafd4a4e6ec libUnrealEditor-Engine.so!USkeletalMesh::CacheDerivedData(FSkinnedAssetCompilationContext*) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:5198]
0x00007cafd4a58e06 libUnrealEditor-Engine.so!USkeletalMesh::ExecutePostLoadInternal(FSkinnedAssetPostLoadContext&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:3757]
0x00007cafd4b8c38a libUnrealEditor-Engine.so!FSkinnedAssetAsyncBuildWorker::DoWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkinnedAssetAsyncCompileUtils.cpp:38]
0x00007cafd2d4f148 libUnrealEditor-Engine.so!FAsyncTaskBase::DoWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:288]
0x00007cafd2d63bb9 libUnrealEditor-Engine.so!FAsyncTaskBase::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:312]
0x00007cafd7af7000 libUnrealEditor-Core.so!FQueuedThreadPoolWrapper::FScheduledWork::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Misc/QueuedThreadPoolWrapper.h:142]
0x00007cafd30752b0 libUnrealEditor-Engine.so!FQueuedThreadPoolWrapper::FScheduledWork::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Misc/QueuedThreadPoolWrapper.h:142]
0x0000000000268007 UnrealEditor!FQueuedLowLevelThreadPool::AddQueuedWork(IQueuedWork*, EQueuedWorkPriority)::'lambda'()::operator()() const [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Misc/QueuedThreadPoolWrapper.h:484]
0x0000000000267841 UnrealEditor!LowLevelTasks::TTaskDelegate<LowLevelTasks::FTask* (bool), 48u>::TTaskDelegateImpl<void LowLevelTasks::FTask::Init<FQueuedLowLevelThreadPool::AddQueuedWork(IQueuedWork*, EQueuedWorkPriority)::'lambda'()>(char16_t const*, LowLevelTasks::ETaskPriority, FQueuedLowLevelThreadPool::AddQueuedWork(IQueuedWork*, EQueuedWorkPriority)::'lambda'()&&, LowLevelTasks::ETaskFlags)::'lambda'(bool), false>::CallAndMove(LowLevelTasks::TTaskDelegate<LowLevelTasks::FTask* (bool), 48u>&, void*, unsigned int, bool) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Async/Fundamental/Task.h:499]
0x00007cafd7680ce6 libUnrealEditor-Core.so!LowLevelTasks::FTask::ExecuteTask() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/P

0x00007caefe356ea2 libUnrealEditor-NaniteBuilder.so!Nanite::BuildDAG(TArray<Nanite::FClusterGroup, TSizedDefaultAllocator<32> >&, TArray<Nanite::FCluster, TSizedDefaultAllocator<32> >&, unsigned int, unsigned int, unsigned int, TBounds<float>&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/NaniteBuilder/Private/ClusterDAG.cpp:126]
0x00007caefe361990 libUnrealEditor-NaniteBuilder.so!Nanite::FBuilderModule::Build(Nanite::FResources&, Nanite::IBuilderModule::FInputMeshData&, Nanite::IBuilderModule::FOutputMeshData*, FMeshNaniteSettings const&, TDelegate<void (bool), FDefaultDelegateUserPolicy>) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/NaniteBuilder/Private/NaniteBuilder.cpp:634]
0x00007cafbe0e729b libUnrealEditor-MeshBuilder.so!FSkeletalMeshBuilder::Build(FSkeletalMeshBuildParameters const&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/MeshBuilder/Private/SkeletalMeshBuilder.cpp:181]
0x00007cafbe0e5d6a libUnrealEditor-MeshBuilder.so!FMeshBuilderModule::BuildSkeletalMesh(FSkeletalMeshBuildParameters const&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Developer/MeshBuilder/Private/MeshBuilderModule.cpp:81]
0x00007cafd4a6768b libUnrealEditor-Engine.so!USkeletalMesh::BuildLODModel(ITargetPlatform const*, int) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:6095]
0x00007cafd4ae3a36 libUnrealEditor-Engine.so!FSkeletalMeshRenderData::Cache(ITargetPlatform const*, USkinnedAsset*, FSkinnedAssetCompilationContext*) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMeshRenderData.cpp:316]
0x00007cafd4a4e6ec libUnrealEditor-Engine.so!USkeletalMesh::CacheDerivedData(FSkinnedAssetCompilationContext*) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:5198]
0x00007cafd4a58e06 libUnrealEditor-Engine.so!USkeletalMesh::ExecutePostLoadInternal(FSkinnedAssetPostLoadContext&) [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkeletalMesh.cpp:3757]
0x00007cafd4b8c38a libUnrealEditor-Engine.so!FSkinnedAssetAsyncBuildWorker::DoWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/SkinnedAssetAsyncCompileUtils.cpp:38]
0x00007cafd2d4f148 libUnrealEditor-Engine.so!FAsyncTaskBase::DoWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:288]
0x00007cafd2d63bb9 libUnrealEditor-Engine.so!FAsyncTaskBase::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:312]
0x00007cafd7af7000 libUnrealEditor-Core.so!FQueuedThreadPoolWrapper::FScheduledWork::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Misc/QueuedThreadPoolWrapper.h:142]
0x00007cafd30752b0 libUnrealEditor-Engine.so!FQueuedThreadPoolWrapper::FScheduledWork::DoThreadedWork() [/home/rv/workspace/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/Misc/QueuedThreadPoolWrapper.h:142]
[... cut ...]
[2025.05.19-03.08.41:885][  0]LogPakFile: Destroying PakPlatformFile
[2025.05.19-03.08.41:885][  0]LogExit: Exiting.
[2025.05.19-03.08.41:885][  0]LogInit: Tearing down SDL.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
^CSegmentation fault (core dumped)
ninja: build stopped: interrupted by user.

**Scripts**
Add any scripts relevant for the reproducibility of the error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

dedoogong, 2025-05-19:
By the way, simple check code like 
```
import carla
import random

client = carla.Client("localhost", 2000)
client.set_timeout(5.0)
world = client.get_world()
map = world.get_map()

spawn_point = random.choice(map.get_spawn_points())
print("Map:", map.name)
print("Spawn Point:", spawn_point)

```
works fine. 
the result is 
```
Map: Carla/Maps/Town10HD_Opt
Spawn Point: Transform(Location(x=-41.825108, y=-6.604221, z=0.600000), Rotation(pitch=0.000000, yaw=-90.161217, roll=0.000000))
```

Blyron, 2025-05-19:
It seems your UE5 was creating internal mesh representation while you launched the manual control. 
You should press play and wait for a bit. Until console doesn't show any "work in progress task"

closed 2025-08-04 by Blyron