carla-simulator/carla #8936: Cannot load Town12

TomoCoder123, 2025-05-23:
I am unable to load Town12 through the client. I am able to load Town12 in the leaderboard script but I cannot load it in by itself. I am on version 9.15 

File "/home/CarlaEndToEnd/simulator.py", line 8, in <module>
    client.load_world('Town12')
RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

GoodarzMehr, 2025-05-25:
Increase the timeout. `client.set_timeout(80)`

TomoCoder123, 2025-05-29:
This solved the timeout issue but still is not able to load in Town 12. I've also tried the suggestion [https://github.com/carla-simulator/carla/issues/6173](url), 
bp=random.choice(world.get_blueprint_library().filter('vehicle'))
bp.set_attribute('role_name', 'hero')
spawn_point = random.choice(world.get_map().get_spawn_points())
ego = world.try_spawn_actor(bp, spawn_point)
with these lines of code, but it did not work.

GoodarzMehr, 2025-05-29:
If you are on 0.9.15 that solution wouldn't make a difference because the spectator act as the hero vehicle and the tiles around it should get loaded.

It can take a minute for Town12 to get loaded. If the timeout issue is resolved then why is it not loaded? If it were not loaded it would time out after 80 seconds.

P.s. when you load Town12 it is usually loaded way above you, have you looked around?

TomoCoder123, 2025-05-30:
Sorry, yes to clarify, it is loaded way above me, how would I be able to spawn it on the ground?

GoodarzMehr, 2025-05-30:
If you want to change the spectator position you can use `spec = world.get_spectator()` and then use `spec.set_location()` to set its location. See the PythonAPI documentation for more details.

Blyron, 2025-06-11:
Please use the provided templates when opening issues. Closing it. Feel free to reopen with a proper template

closed 2025-06-11 by Blyron