carla-simulator/carla #8953: How to use left_hand_traffic

ren-takahara, 2025-06-02:
I don't understand how to use the left-hand driving feature in the left_hand_traffic branch.
I'm trying to create a custom map in RoadRunner, import it, and use it, but could you tell me if there is any information or steps I need to take to make it compatible with left-hand driving?

glopezdiest, 2025-06-02:
Hey @ren-takahara, before answering that the Left Hand traffic feature is now official, and you can use our dev versions (in both UE4 and UE5). We are not really experts in RoadRunner so there might be another way but I'll explain how we tested it:

First of all **create the road** like normal. Then **reverse the lane directions**. This is done not because of the direction itself but because of the intersections. When you create traffic lights in Roadrunner, they are automatically assigned to the intersection entrances. As such, the entrances have to be LHT for the traffic lights to be assigned to the correct lane.

**Move the traffic light to the left side of the road**, as they are generally created on the right, following RHT conventions. Then **add the road traffic elements** (speed limit, etc) on the left side of the road. 

**Export the map**. Choose either RHT or LHT. From the [OpenDrive guidelines](https://publications.pages.asam.net/standards/ASAM_OpenDRIVE/ASAM_OpenDRIVE_Specification/latest/specification/10_roads/10_01_introduction.html), LHT is defined as an attribute of the road, and neither the RHT or LHT options in RoadRunner add that.

Manually **add the LHT rule**, you can use this python script: [add_lht_rule.zip](https://github.com/user-attachments/files/20551200/add_lht_rule.zip) (this step is not needed if you are using the branch).

And that's it, choose any of the CARLA pipelines available like normal (import the map, use the opendrive standalone...)

ren-takahara, 2025-06-05:
After building with python3.8, I was able to run generate_traffic.py.
However, the spawned agents are driving on the right side.
I have added LHT rules to OpenDrive.
Is there still a step I need to take?

ren-takahara, 2025-06-05:
The path to the python API was incorrect.
I've solved it so I'll close it. Thank you.