carla-simulator/carla #9019: Request to Contribute – Software Engineer Interested in CARLA

johanz9, 2025-07-01:
Hi CARLA team,

I'm a software engineer with several years of experience developing ERP systems and business management tools. Recently, I've become deeply interested in the field of autonomous driving and, while researching, I came across the CARLA simulator.

I'm really impressed by the quality and openness of the project. I would love to contribute or get involved in any way I can — whether that's testing experimental tools, reporting bugs, or collaborating on development. Please let me know how I can best participate and support the community.

Thank you for your amazing work and for keeping CARLA open-source and community-driven.

Best regards,
Johan Manuel Mendoza Coronado

Blyron, 2025-07-01:
We are open to contributions. You can try to fix issues community are having just by looking at issues

Blyron, 2025-07-02:
You could start here https://github.com/carla-simulator/carla/issues/9022

closed 2025-07-16 by Blyron