carla-simulator/carla #9029: Build Failure in CARLA on Ubuntu 24.04 Due to Clang and libc Incompatibility

hh-xiaohu, 2025-07-05:
When I built carla on Linux, I encountered the following error:

`(carla) usr@usr-ubt:~/Workspace/carla/carla_dev$ make launch
Util/BuildTools/Linux.mk:163: warning: overriding recipe for target 'downloadplugins'
Util/BuildTools/Linux.mk:142: warning: ignoring old recipe for target 'downloadplugins'
Setup.sh: boost-1.80.0-c10 already installed.
Setup.sh: rpclib-v2.2.1_c5-c10 already installed.
Setup.sh: gtest-1.8.1-c10 already installed.
Setup.sh: recast-c10 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Eigen already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 4457247-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildOSM2ODR.sh: Building OSM2ODR.
/home/usr/Workspace/carla/carla_dev/Build/libosm2dr-source
CMake Deprecation Warning:
  Support for "Extra Generators" like

    Eclipse CDT4

  is deprecated and will be removed from a future version of CMake.  IDEs may
  use the cmake-file-api(7) to view CMake-generated project build trees.

CMake Warning (dev) at CMakeLists.txt:33 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is Clang 10.0.1
-- The CXX compiler identification is Clang 10.0.1
-- Eclipse version is set to 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/bin/clang
**-- Check for working C compiler: /usr/bin/clang - broken
CMake Error at /usr/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/usr/bin/clang"**

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/home/usr/Workspace/carla/carla_dev/Build/libosm2dr-build/CMakeFiles/CMakeScratch/TryCompile-UEErww'
    
    Run Build Command(s): /usr/bin/ninja -v cmTC_5c855
    [1/2] /usr/bin/clang    -MD -MT CMakeFiles/cmTC_5c855.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_5c855.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_5c855.dir/testCCompiler.c.o -c /home/usr/Workspace/carla/carla_dev/Build/libosm2dr-build/CMakeFiles/CMakeScratch/TryCompile-UEErww/testCCompiler.c
    [2/2] : && /usr/bin/clang  -rdynamic CMakeFiles/cmTC_5c855.dir/testCCompiler.c.o -o cmTC_5c855   && :
    FAILED: cmTC_5c855 
    : && /usr/bin/clang  -rdynamic CMakeFiles/cmTC_5c855.dir/testCCompiler.c.o -o cmTC_5c855   && :
    /usr/bin/ld: warning: /usr/bin/../lib/gcc/x86_64-linux-gnu/13/../../../x86_64-linux-gnu/crt1.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
    /usr/bin/ld: warning: /lib/x86_64-linux-gnu/libc.so.6: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
    /usr/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: unknown type [0x13] section `.relr.dyn'
    /usr/bin/ld: skipping incompatible /lib/x86_64-linux-gnu/libc.so.6 when searching for /lib/x86_64-linux-gnu/libc.so.6
    /usr/bin/ld: cannot find /lib/x86_64-linux-gnu/libc.so.6
    /usr/bin/ld: /lib64/ld-linux-x86-64.so.2: unknown type [0x13] section `.relr.dyn'
    /usr/bin/ld: skipping incompatible /lib64/ld-linux-x86-64.so.2 when searching for /lib64/ld-linux-x86-64.so.2
    **/usr/bin/ld: cannot find /lib64/ld-linux-x86-64.so.2
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.**
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:33 (project)

-- Configuring incomplete, errors occurred!
make: *** [Util/BuildTools/Linux.mk:157: osm2odr] Error 1`

It seems that there might be compatibility issues between the environments. My system version: 

1. Ubuntu 24.04
2. GNU ld (GNU Binutils) 2.31.1
3. GNU C Library (Ubuntu GLIBC 2.39-0ubuntu8.4) stable release version 2.39
4. Ubuntu clang version 10.0.1-++20211003085942+ef32c611aa21-1exp120211003090334.2
5. GCC version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04).

I have tried various environment combinations, such as changing binutils to 2.42 and clang to 18, but I still encounter the same error. I would like to ask if anyone has successfully configured this on Ubuntu 24.04, as I would like to inquire about the correct environment setup.

shubhamKGIT, 2025-08-11:
I tried to solve the clang compatibility then moved to docker builds recently. Try that.

MarcelPiNacy-CVC, 2025-08-20:
Hi, Ubuntu 24 is currently unsupported, sorry. Move to 22 or you will have to wait for us to add the changes. Alternatively, do as @shubhamKGIT said.