carla-simulator/carla #9067: Carla server appears to crash when launched repeatedly

rzalawad, 2025-07-22:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: Docker container with base image carlasim/carla:0.9.15. Host OS: Ubuntu 22.04.5 LTS
  * Python version: 3.7
  * GPU: 4 NVIDIA A10G
  * GPU Drivers: 550.144.03

**Describe the bug**
When the carla simulator is run repeatedly (1 instance / GPU on 1 GPU only), then, non-deterministically, one or more carla instances crash with the following error
```
sh: 1: xdg-user-dir: not found
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 1214]
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Enough GPU memory is available as I have a dedicated instance with nothing running on the GPU.

**Steps to reproduce**
1. Download the docker carlasim/carla:0.9.15 docker container
2. Create a workdir folder and place the python code in a file named `carla_crash_test.py`
3. Run the given bash command.

The script launches carla (in a docker container) iteratively N number of times and reports if carla has crashes by examining the logs of carla.

Command on host
```bash
python3 workdir/carla_crash_test.py 10 --num-gpus 1 --network-level 0
```
Please refer to the --help section to see what different arguments mean

**Expected behavior**
Regardless of the number of times the simulator is run, the simulator should not crash if enough memory is available.

**Logs**
Paste the server and client logs. Server ones can be found at:
  * Unreal Engine 4: _Unreal/CarlaUE4/Saved/Logs_
**How do i get these? Adding -Log to the carla launch command does nothing**

**Scripts**

<details><summary>Code</summary>
<p>

```python
import os
import subprocess
import time
import sys
import argparse
from uuid import uuid4
from datetime import datetime
import concurrent.futures
import threading

def launch_carla_container(iteration, gpu_id, network_level, use_ipc_host):
    """Launch a single CARLA container and return container name"""
    name = uuid4().hex[:8]
    container_name = f"carla_test_{iteration}_gpu{gpu_id}_{name}"
    
    cmd = [
        "docker",
        "run",
[... 350 lines cut ...]
            iter_results = [r for r in all_results if r['iteration'] == iteration and r['container_name']]
            iter_crashes = sum(1 for r in iter_results if r['crashed'])
            iter_total = len(iter_results)
            if iter_total > 0:
                f.write(f"  Iteration {iteration}: {iter_crashes}/{iter_total} crashed ({(iter_crashes/iter_total)*100:.1f}%)\n")
        
        f.write("\nDetailed crash information:\n")
        if crashed_results:
            for result in crashed_results:
                f.write(f"  - Iteration {result['iteration']}, GPU {result['gpu_id']}: {result['fault_type']} "
                        f"(Container: {result['container_name']}, Log: {result['log_file']})\n")
        else:
            f.write("  No crashes detected!\n")
    
    print(f"\nSummary saved to: {summary_file}")

if __name__ == "__main__":
    main() 
```

</p>
</details> 

Issue is different from https://github.com/carla-simulator/carla/issues/7597 as no logical code (changes in simulator) are executed (actor creation etc.)
Issue https://github.com/carla-simulator/carla/issues/7710 might be similar but doesn't provide a reproducible

vit-hamin, 2025-07-22:
I think this is due to the memory leak issue in carla when running the same script multiple times.
Even if you clean or destroy every actor, carla seems to not release all of its memory resources,
and as a result, if you run it multiple times, it raises a segmentation fault (trying to access memory it's not allocated to).

On the client, if you press ~(tilda) and enter 'stat_rhi' you can see that the memory buffer will slightly increase with each script run.

It's a client-side error, there is little you can do to solve this problem exclusively using the script. All we can do as of now is restart the client every once in a while.

rzalawad, 2025-07-23:
@vit-hamin Thank you for your response. I might be asking some beginner questions so forgive me.

1. To run the stat_rhi command, would I need to launch the game with a UI (no -RenderOffScreen and a display attached)?
2. I am not running any script nor do I have a client connecting to the server in the example code. When carla is launched, does a default map with default actors get loaded? If yes, then your explanation makes sense. If not, then maybe this is another issue.
3. If we do have a memory leak, on what resource do you think we have one? CPU or GPU memory? I'm asking because in either case, I have ample memory available (I monitor the nvidia-smi and htop utilization in another tmux pane.) Since I never reach close to the max (the machine has 187GB of CPU memory,) it is perplexing how a malloc fails (my assumption being that we try to read or write to the return value of malloc which causes the segmentation fault).

In parallel, I'll try to create a debug build of the game to see where the segmentation fault might be occurring. Any guidance is much appreciated :)

rzalawad, 2025-08-11:
Hello @Rak-r This might be the wrong issue for your error since in this thread, the focus is identifying why carla crashes without any interaction from any client.

rzalawad, 2025-08-14:
When running the carlaUE4.sh script repeatedly, I am able to reproduce the reported error on 0.9.15 but not on the nightly carla release build on the ue4-dev branch (as of yesterday.)

I am still able to reproduce the segmentation fault (with error `Malloc Size=65538 LargeMemoryPoolOffset=65554`) when launching carla on 2 different gpus simultaneously on the latest nightly release (I believe commit `b081c211484df233c8064b74d2ddab8d47488cc4`). 
Considering this, I will close the current issue since repeated launches on 1 gpu with no clients connected may only be raising a segmentation fault on 0.9.15.
Thank you to everyone who read the tried to help :)

closed 2025-08-14 by rzalawad

Rak-r, 2025-08-14:
Hello @rzalawad , could you provide some insight on my issue as well. I am a bit new to Carla and struggling to find why when I call Ctrl+c on my python script the server crashes after few seconds. Could you check the script running on your setup? It would be helpful if someone can guide.