carla-simulator/carla #9098: Missing vehicle labels

Luiz-Puma, 2025-07-30:
**Setup**
* CARLA version: 0.10.0
* Platform: Ubuntu 22.04
* Python version: 3.10.12
* GPU: RTX 5070 Ti
* GPU Drivers: 570.153.02 (Latest Production)

**Describe the bug**
I am trying to import new vehicles to CARLA but some labels are missing, like instance segmentation and blueprint name.

**Steps to reproduce**
Import .fbx vehicle, create vehicle blueprint and place on scene.

**Expected behavior**
The vehicle are visible in instance segmentation and listed in blueprint library.

**Screenshots**
[image]
[image]

Blyron, 2025-07-30:
Enable Nanite in those meshes, Remember nanite cannot be enabled in meshes with transparency

Luiz-Puma, 2025-09-09:
I find a solution editing the vehicles configuration file "CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Config/VehicleParameters.json", just add my vehicle in the file.

{
	"Make": "motorcycle",
	"Model": "motorbike1",
	"Class": "/Game/Carla/Blueprints/Vehicles/MotorBike1/BP\_MotorBike1.BP\_MotorBike1_C",
	"NumberOfWheels": 2,
	"Generation": 3,
	"ObjectType": "",
	"BaseType": "motorcycle",
	"SpecialType": "",
	"HasDynamicDoors": true,
	"HasLights": true,
	"RecommendedColors": [
		{
			"R": 232,
			"G": 192,
			"B": 0,
			"A": 255
		}
	],
	"SupportedDrivers": []
}

The images:
RGB
[image]
Instance Segmentation
[image]

closed 2025-09-09 by Luiz-Puma