carla-simulator/carla #9118: No Crosswalks Returned by the Map

GoodarzMehr, 2025-08-05:
**Setup**
  * CARLA version: 0.9.15
  * Platform: Ubuntu 22.04
  * Python version: 3.10.12
  * GPU: 2xRTX 4090s
  * GPU Drivers: 550.163.01

**Describe the bug**
In the `Town12`, `Town13`, and `Town15` maps, using `maps.get_crosswalks()` returns an empty list.

**Solution**
In the OpenDRIVE `.xodr` files of these maps, crosswalk objects (CContinentalCrosswalk) have `type="-1"`. Changing that to `type="crosswalk"` seems to resolve the issue.

Blyron, 2025-08-06:
Having a look to have it fixed before 0.9.16 release

GoodarzMehr, 2025-08-06:
Thanks. I used this to make the changes:
```Python
import xml.etree.ElementTree as ET

tree = ET.parse('./Town12.xodr')
root = tree.getroot()

for road in root.findall('road'):
    if road.find('objects') is not None:
        objects = road.find('objects').findall('object')
        for obj in objects:
            if obj.get('name') == 'CContinentalCrosswalk':
                obj.set('type', 'crosswalk')

tree.write('./Town12.xodr', encoding='utf-8', xml_declaration=True)
```

BTW, any plans to address #5790 ?

Blyron, 2025-08-06:
> Thanks. I used this to make the changes:
> 
> import xml.etree.ElementTree as ET
> 
> tree = ET.parse('./Town12.xodr')
> root = tree.getroot()
> 
> for road in root.findall('road'):
>     if road.find('objects') is not None:
>         objects = road.find('objects').findall('object')
>         for obj in objects:
>             if obj.get('name') == 'CContinentalCrosswalk':
>                 obj.set('type', 'crosswalk')
> 
> tree.write('./Town12.xodr', encoding='utf-8', xml_declaration=True)
> BTW, any plans to address [#5790](https://github.com/carla-simulator/carla/issues/5790) ?

We are taking other route not modifying Xodrs. moreover, your solution is incomplete because there are other crosswalks with other names. We are planning to fix it, but we do not know if its going to make to the next release