carla-simulator/carla #9125: Carla 0.9.15 directly exit without any error!

HuangChiEn, 2025-08-07:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.15
  * Platform: Ubuntu 22.04.4 LTS
  * Python version: 3.10.15
  * GPU: H100
  * GPU Drivers: 535.161.08

**Describe the bug**
As screen-shot, Carla shell directly exit without given any error..
[image]

**Steps to reproduce**
`/path_to_carla/CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=38400 -graphicsadapt &`
I have checked by `ps -aux`, it also not appear in background process!!

**Expected behavior**
A carla server running on background process. 

**Logs**
Paste the server and client logs. Server ones can be found at:
? Doesn't exists any log in working dir..

**Scripts**
None

**Screenshots**
[image]

HuangChiEn, 2025-08-19:
> Hi, I don't understand what the issue is. Is the server immediately exiting?

yes, as you can saw in the screen shot, carla process is no longer run in background (completely exit).

HuangChiEn, 2025-08-19:
meta info : 
Since community suggest precompile version. As i known there're 4 way to install CARLA : 
1. "CARLA in docker", pulling docker image only described in document 0.9.12 and **disappear** in post version (0.9.14, 0.9.15, and so on), not sure it's still maintained (or a recommended way) ? In addition, one of server i can use is **container**, so docker deamon even not exists! 
> BTW, i don't want to study how to build "docker in docker")  
2. "Install debin package", while this solution placed in all version document, **the debin package distributor server** is **down for several years** [see issue](https://github.com/carla-simulator/carla/issues/6326#issue-1633528858).
> server in 2025/8 is still down...( had tried it last week
3. "Install precompiled version", now i'm going this way, cause i follow this [commend](https://github.com/RenzKa/simlingo/tree/main/Bench2Drive#setup) to download CARLA and directly run CARLA commend. 
4. "Build from source", forgive me i haven't tried it before. Seems to challenge...orz

For going route 3, precompiled version have some issue in my case as above. hope those info help!

> > Hi, I don't understand what the issue is. Is the server immediately exiting?
> 
> yes, as you can saw in the screen shot, carla process is no longer run in background (completely exit).

> Hi, I don't understand what the issue is. Is the server immediately exiting?

HuangChiEn, 2025-08-19:
> Hi, I don't understand what the issue is. Is the server immediately exiting?

MarcelPiNacy-CVC, 2025-08-20:
I have manually downloaded the package you pointed to and it seems to run fine. Could you share the output of `nvidia-smi` and `vulkaninfo`? If `vulkaninfo` is not available install it with `sudo apt install vulkan-tools`.
Regarding point 2, we have deprecated debian packages.

HuangChiEn, 2025-08-21:
> I have manually downloaded the package you pointed to and it seems to run fine. Could you share the output of `nvidia-smi` and `vulkaninfo`? If `vulkaninfo` is not available install it with `sudo apt install vulkan-tools`. Regarding point 2, we have deprecated debian packages.

Nvidia Driver :
[image]

exit immediately (precompile version 0.9.15)
[image]

vulkaninfo is impossible installed in the server (server prohibit adding new apt source)
> for ubuntu, vulkaninfo require adding new apt source