carla-simulator/carla #9213: Dark Pygame in manual_control.py Build Version

DavidMagyarics, 2025-08-31:
Setup

CARLA version: 0.10.0 (built from source)

Platform: Windows 11

Python version: 3.8.10

Unreal Engine version: 5.5 (CarlaUE5 build)

GPU: RTX 5070

Describe the bug
When I run the manual_control.py script, the simulation appears extremely dark. The build process for CARLA 0.10.0 completed successfully, and the simulator runs, but the rendering brightness is much lower than expected, making it very difficult to see the environment.

Expected behavior
The scene should render with normal lighting when running manual_control.py, instead of appearing overly dark.

Picture

[image]

Thank you for any potential responses.

Blyron, 2025-08-31:
If you pull latest changes to code and content. you will have this fixed.

atasoglou, 2025-09-25:
> I suppose it connects with settings for the rgb camera. Unfortunately there aren't many attributes in the ue5-dev branch. I managed it when built package on commit `ce6eb585aac4d1278b0cfa911fc3942b8eb5b3e9`. Also the next settings for rgb camera help me to make a beautiful image: `camera_bp.set_attribute("fstop", "16.0")` `camera_bp.set_attribute("iso", "100000.0")` `camera_bp.set_attribute("gamma", "2.2")` `camera_bp.set_attribute("temp", "5500.0")` `camera_bp.set_attribute("toe", "0.55")` `camera_bp.set_attribute("slope", "0.88")` `camera_bp.set_attribute("tint", "0.0")` `camera_bp.set_attribute("lens_flare_intensity", "0.0")` `camera_bp.set_attribute("shutter_speed", "200.0")` `camera_bp.set_attribute("bloom_intensity", "0.675")` `camera_bp.set_attribute("focal_distance", "1000.0")` `camera_bp.set_attribute("exposure_compensation", "7.5")` `camera_bp.set_attribute("exposure_max_bright", "12.0")` `camera_bp.set_attribute("motion_blur_intensity", "0.0")` `camera_bp.set_attribute("chromatic_aberration_intensity", "0.0")`

Thanks for the suggestion! I tried adding those in the existing [`manual_control.py`](https://github.com/carla-simulator/carla/blob/ue5-dev/PythonAPI/examples/manual_control.py)  like this:
```
bp.set_attribute("fstop", "16.0")
bp.set_attribute("iso", "100000.0")
bp.set_attribute("gamma", "2.2")
bp.set_attribute("temp", "5500.0")
bp.set_attribute("toe", "0.55")
bp.set_attribute("slope", "0.88")
bp.set_attribute("tint", "0.0")
bp.set_attribute("lens_flare_intensity", "0.0")
bp.set_attribute("shutter_speed", "200.0")
bp.set_attribute("bloom_intensity", "0.675")
bp.set_attribute("focal_distance", "1000.0")
bp.set_attribute("exposure_compensation", "7.5")
bp.set_attribute("exposure_max_bright", "12.0")
bp.set_attribute("motion_blur_intensity", "0.0")
bp.set_attribute("chromatic_aberration_intensity", "0.0")
```

I get for all of these error that the attribute does not exist. Even more interesting is that [the existing code](https://github.com/carla-simulator/carla/blob/ue5-dev/PythonAPI/examples/manual_control.py#L1150-L1151):
```
                if bp.has_attribute('gamma'):
                    bp.set_attribute('gamma', str(gamma_correction))
```
actually also never runs for me. So the `--gamma` argument you can set while running it, it is also not doing anything.

So I am now confused: Is it an UE5 issue or just the way the pygame is setup? 🧐🤔

atasoglou, 2025-09-25:
Actually running this for the RGB camera
```python
for attr in bp:
    print(f"- {attr.id}: {attr.type} (default: {attr.recommended_values if attr.recommended_values else 'N/A'})")
```
 I get:
```
- enable_postprocess_effects: Bool (default: ['true'])
- lens_x_size: Float (default: ['0.08'])
- lens_k: Float (default: ['-1.0'])
- lens_kcube: Float (default: ['0.0'])
- lens_circle_multiplier: Float (default: ['0.0'])
- post_process_profile: String (default: ['default'])
- fov: Float (default: ['90.0'])
- image_size_x: Int (default: ['800'])
- lens_y_size: Float (default: ['0.08'])
```

closed 2026-11-12 by Blyron