carla-simulator/carla #9234: Failed to import a prop，please help me.

nobub, 2025-09-05:
**Setup**
 the setup i'm using to run CARLA along with its version:
  * CARLA version: 0.9.14
  * Platform: WIndows
  * Python version: 3.10

**Describe the bug**
I'm trying to import a static component. I have created the relevant blueprint class and added it to PropFactory, but I can't find the component I imported in the blueprint library.

**Steps to reproduce**
I imported the fbx file into carla and created a blueprint class that inherited the Actor for it. Then I added it to the factory, but no matter which factory it was added to, it couldn't be found in the blueprint.

**Expected behavior**
I hope to import a prop and control it through the Python API.

**Screenshots**

[image]

[image]

[image]

Blyron, 2025-09-08:
We do not support Blueprints spawnning.

https://carla.readthedocs.io/en/latest/content_authoring_props/
Here how to spawn a prop.

closed 2025-09-08 by Blyron

BaronBunny427, 2025-09-20:
It works, thanks.