carla-simulator/carla #9278: Carla 0.9.16 Ubunto ROS2 topic with double slash

FriedrichZimmer, 2025-09-16:
**Setup**
Describe the setup you are using to run CARLA along with its version:
  * CARLA version: 0.9.16 Release version
  * Platform: Ubuntu 22.04
  * Python version: 3.10.12
  * GPU: Nvidia Quadro P4000
  * GPU Drivers: Don't know

**Describe the bug**
When attaching a ROS2 camera to a vehicle the published topic is having a double slash // in its name.
Because of the double slash I cannot visualize those images in rviz, lichtblick and "ros2 topic echo".

**Steps to reproduce**
I spawned a vehicle in town and attached a RGB camera and enabled ROS with
camera_bp = blueprint_library.find('sensor.camera.rgb')
camera = world.spawn_actor(camera_bp, transform, attach_to=vehicle)
camera.enable_for_ros()

with "ros2 topic list" I can see that the following two new topics have been created:
/carla//actor107/camera_info
/carla//actor107/image

in the command name the following command:
"ros2 topic echo /carla//actor107/image"
fails with the following error:
xmlrpc.client.Fault: <Fault 1: "<class 'rclpy.exceptions.InvalidTopicNameException'>:Invalid topic name: topic name must not contain repeated '/':\n  '/carla//actor107/image'\n          ^">

I renamed the topic with:
camera_bp.set_attribute('ros_name', '01_default_new')
but this just resalted in the same error:
/carla//01_default_new/camera_info
/carla//01_default_new/image

**Expected behavior**
The ROS2 topic names should be:
/carla/actor107/camera_info
/carla/actor107/image

**Logs**
Not found in the specified location.

**Scripts**
See above

**Screenshots**
[image]

liuxinyiwssy, 2025-09-19:
Hello, I'm trying the native ROS2 code in the example, but no topics are being sent. Is there something I'm doing wrong? I followed the documentation and started Carla with `make` launch ARGS="--ros2 --editor-flags='--ros2'"` and then `python3 ros2_native.py --file stack.json` , but I don't see any topics after `ros2 topic list`.

FriedrichZimmer, 2025-10-02:
When playing around with ros2_native.py in the PythonAPI examples, I noticed that everything is fine as long as the vehicle id is "hero". The issue only occurs with custom names.

M-Colley, 2026-10-26:
Draft PR #10026 proposes a `ue4-dev` fix for the doubled slash when a ROS 2 publisher has an empty parent name, with DDS-free topic-name regressions. It currently has merge conflicts. Live ROS 2 publisher/subscriber validation remains unrun.