carla-simulator/carla #9294: Carla 0.9.16 ROS2 Native: support alternative DDS implementations

maxime-clem, 2025-09-20:
**Is your feature request related to a problem? Please describe.**
The current ROS2 native implementation for Carla 0.9.16 only supports eProsima’s Fast DDS.

**Describe the solution you'd like**
The DDS used by Carla 0.9.16 can be configured and different implementations can be used.

**Describe alternatives you've considered**
Changing the DDS implementation used on my system.

**Additional context**
ROS2 allows using [alternative DDS implementations](https://docs.ros.org/en/humble/Installation/RMW-Implementations/DDS-Implementations.html). For example [CycloneDDS](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/#cyclonedds-configuration) is used in Autoware.

ali-0-1, 2025-09-30:
Same is true for **Carla 0.10.0**.
Due to fastdds implementation we could not get topics from the vehicle which is using cyclonedds. Carla is dominating the communication, exporting cyclonedds also has no effect.
Is there any work around or implementation that can solve the issue?
For now, we used another machine in the same network which can access pyhtonApi of Carla and get data from the vehicle, that is the work around we found for now.

We have Ros2 humble installed in both machines and the vehicle.

ali-0-1, 2025-10-07:
**UPDATE:**
Actually our case was not related to implementation type but network configuration and firewall settings. While the Carla and vehicle in the same network with proper settings they can communicate without additional implementation change. 
Vehicle has cyclonedds implementation and Carla by default Fastdds.

JesusAnaya, 2026-03-11:
@maxime-clem I have been working for a while on implementing CycloneDDS in Carla UE4. I think I'm very close to getting a stable version. Furthermore, I will prepare a PR with these changes to ue4-dev very soon. Additionally, this refactoring includes a new C++ interface in the Carla library to implement other DDS backends in a more structured and "easy" way. Probably at the end of this week, I can have it ready.

JesusAnaya, 2026-03-16:
Added PR, you can clone my branch and compile from it for testing.